luigifreda/pyslam #2: attribute error related to g2o

naughtyStark, 2020-01-14:
Hi, I am quite new to computer vision based localization and just wanted to try this out. When running the main_slam.py I get the following error.
```
  File ".../pyslam/optimizer_g2o.py", line 96, in optimization
    edge.fx = f.fx 
AttributeError: 'g2o.EdgeSE3ProjectXYZ' object has no attribute 'fx'
```
I understand that this error is related to g2opy, but I was wondering how everyone else got around this, since the above object has not been used in any of the examples given in g2opy. It would also be helpful if there was another way to obtain whatever is required here (as in, if there is a member function of that object from which I should get the values).

P.S: I made no changes to the source code of pyslam or g2o

naughtyStark, 2020-01-14:
I figured out this particular error. You have included a replacement file in pyslam/lib/g2opy_changes
```
types_six_dof_expmap.h
```
For anyone else getting this error, you have to replace the file in
```
g2opy/python/types/sba/
```
and then install g2o (assuming you have already created the build directory)
```
cmake ..
make -j8
cd ..
python3 setup.py install
```
@luigifreda please include these instructions in the readme.

luigifreda, 2020-05-03:
Hi,
the error you experienced was due to the fact that you did not use the provided installation script. Please, read the install instructions. 

In any case, check the new pySLAM v2. New features and new instructions.

closed 2020-05-09 by luigifreda

robovis, 2020-05-22:
Even in the modified files, the EdgeStereoSE3ProjectXYZOnlyPose() is not fully included (attributes like fx, fy, ...). I suppose adding a few lines in the types_six_dof_expmap.h should solve the problem.

robovis, 2020-05-22:
Here is the updated file:
[types_six_dof_expmap.zip](https://github.com/luigifreda/pyslam/files/4669648/types_six_dof_expmap.zip)

luigifreda, 2020-05-22:
Hi, 
thanks for your note. Everything that is needed is in this file 
https://github.com/luigifreda/pyslam/blob/master/thirdparty/g2opy_changes/types_six_dof_expmap.h 
If you have troubles, as explained [here](https://github.com/luigifreda/pyslam/blob/master/TROUBLESHOOTING.md#user-content-g2o-error), the reason is that you did not use the required install script 
https://github.com/luigifreda/pyslam/blob/master/install_thirdparty.sh 
Please, read the install instructions and take a glance at this [file](https://github.com/luigifreda/pyslam/blob/master/TROUBLESHOOTING.md#user-content-g2o-error) too.

robovis, 2020-05-22:
I've already looked at the files you mentioned. The problem is that the public attributes you've added are only for the "EdgeSE3ProjectXYZ" and "EdgeSE3ProjectXYZOnlyPose" edges. You've left out public methods (like fx, fy, cx, cy, bf) for "EdgeStereoSE3ProjectXYZ" and "EdgeStereoSE3ProjectXYZOnlyPose". 

BTW I'm writing an ORB-SLAM2 equivalent in python using g2o and pyDBoW. Would that be useful for your repo?

YJonmo, 2023-08-22:
> I've already looked at the files you mentioned. The problem is that the public attributes you've added are only for the "EdgeSE3ProjectXYZ" and "EdgeSE3ProjectXYZOnlyPose" edges. You've left out [...]

Hi there,

Do you have a link to your ORB-SLAM2 equivalent in python?