luigifreda/pyslam #3: missing instructions

naughtyStark, 2020-01-14:
I found a couple of instructions missing from the readme so I am writing them here, @luigifreda please include them in the readme so that newbies like me don't stay stuck with these errors: 
## 1) SIFT error:
While this is common knowledge to most people that have worked with opencv, for a newcomer it may not be, so I have included the fix here anyway: 
(Assuming you are running python>=3.5)
```
pip3 install opencv-contrib-python==3.4.0.12
```
this should fix the error.

## 2) g2o Error: 
``` 
 File ".../pyslam/optimizer_g2o.py", line 96, in optimization
    edge.fx = f.fx 
AttributeError: 'g2o.EdgeSE3ProjectXYZ' object has no attribute 'fx'
```
fix: 
There is a replacement file in pyslam/lib/g2opy_changes
```
types_six_dof_expmap.h
```
you have to replace the file in
```
g2opy/python/types/sba/
```
and then install g2o (assuming you have already created the build directory)
```
cmake ..
make -j8
cd ..
python3 setup.py install
```
## 3) ndarray is not C-contiguous
an error as follows will pop-up:
```
ValueError: ndarray is not C-contiguous
```
DO NOT uninstall/ downgrade numpy to below 1.10.x! The fix for this is simple:
open pyslam/search_points.py, find this line (for me it was line 79): 
```
projs = f_cur.project_map_points(points)
```
and replace it with : 
```
projs = f_cur.project_map_points(points)
projs = projs.copy(order='C')
```
(or just add the second line. I use "replace" so that newbies can just copy-paste it).

LoveMeWithoutAll, 2020-02-02:
Thank you for your guide

luigifreda, 2020-05-03:
Hi,
thanks for your suggestions. 
Check the new pySLAM v2: new features and new instructions!

closed 2020-05-09 by luigifreda