luigifreda/pyslam #11: Using SIFT

alexs7, 2020-05-09:
@luigifreda Thanks for this great work!

I am trying to use `tracker_config = FeatureTrackerConfigs.SIFT` in `main_vo.py`

I have noticed that this will create a `DescriptorFeatureTracker` that uses` DES_BF    = 1   # descriptor-based, brute force matching with knn` right ?

My question is about the matcher:

`self.matcher = feature_matcher_factory(norm_type=self.norm_type, ratio_test=match_ratio_test, type=self.matching_algo)`

it seems that `norm_type=self.norm_type` is always set to `norm_type=cv2.NORM_HAMMING`
but for SIFT it should be `NORM_L2`?

luigifreda, 2020-05-09:
Hi,
glad you like pySLAM. 
The norm type is managed [here](https://github.com/luigifreda/pyslam/blob/master/feature_manager.py#L644).
With SIFT, it is set to cv2.NORM_L2 (which is equal to 4) and NOT to cv2.NORM_HAMMING (which is equal to 6). You can test it with a simple print. 
In particular, in the class [FeatureInfo](https://github.com/luigifreda/pyslam/blob/master/feature_types.py#L94), each feature is assigned its correct norm. 
L

alexs7, 2020-05-09:
Oh ok now it makes sense `DescriptorFeatureTracker` calls `feature_manager_factory()` which in turn creates a `FeatureManager` and in there the `norm_type` is set !

Ok thanks!

closed 2020-05-09 by alexs7