luigifreda/pyslam #12: anaconda env create falied

zhangjun-xyz, 2020-05-11:
Hello, I am eager to run pyslam on my Mac. But when I created the anaconda environment on it, I got the following errors:
./pyenv-conda-create.sh
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound:
  - numpy==1.18.1=py36h7314795_1
  - sqlite==3.31.1=h7b6447c_0
  - _tflow_select==2.1.0=gpu
  - tensorboard==1.14.0=py36hf484d3e_0
  - tensorflow-gpu==1.14.0=h0d30ee6_0
  - gnutls==3.6.5=hd3a4fd2_1002
  - gmp==6.2.0=he1b5a44_2
  - gst-plugins-base==1.14.5=h0935bb2_2
  - xorg-libxrender==0.9.10=h516909a_1002
  - pcre==8.44=he1b5a44_0
  - xorg-xextproto==7.3.0=h14c3975_1002
  - cudnn==7.6.5=cuda10.1_0
  - zlib==1.2.11=h7b6447c_3
  - xorg-xproto==7.0.31=h14c3975_1007
  - xorg-libxdmcp==1.1.3=h516909a_0
  - ld_impl_linux-64==2.33.1=h53a641e_7
  - hdf5==1.10.2=hba1933b_1
  - cupti==10.1.168=0
  - jpeg==9c=h14c3975_1001
  - h5py==2.8.0=py36h989c5e5_3
  - tk==8.6.8=hbc83047_0
  - libopenblas==0.3.7=h5ec1e0e_6
  - libffi==3.2.1=hd88cf55_4
  - cudatoolkit==10.1.243=h6bb024c_0
  - protobuf==3.11.4=py36he6710b0_0
  - libgfortran-ng==7.3.0=hdf63c60_5
  - expat==2.2.9=he1b5a44_2
  - xorg-kbproto==1.0.7=h14c3975_1002
  - readline==7.0=h7b6447c_5
  - qt==5.12.5=hd8c4c69_1
  - tensorflow-base==1.14.0=gpu_py36he45bfe2_0
  - openh264==1.8.0=hdbcaa40_1000
  - c-ares==1.15.0=h7b6447c_1001
  - fontconfig==2.13.1=h86ecdb6_1001
  - zstd==1.4.4=h3b9ef0a_2
  - libclang==9.0.1=default_hde54327_0
  - gstreamer==1.14.5=h36ae1b5_2
  - nss==3.47=he751ad9_0
  - dbus==1.13.6=he372182_0
  - xz==5.2.4=h14c3975_4
  - openssl==1.1.1e=h7b6447c_0
  - wrapt==1.12.1=py36h7b6447c_1
  - libtiff==4.1.0=hc3755c2_3
  - xorg-libxau==1.0.9=h14c3975_0
  - pixman==0.38.0=h516909a_1003
  - libuuid==2.32.1=h14c3975_1000
  - harfbuzz==2.4.0=h9f30f68_3
  - lz4-c==1.8.3=he1b5a44_1001
  - freetype==2.10.1=he06d7ca_0
  - libxcb==1.13=h14c3975_1002
  - graphite2==1.3.13=he1b5a44_1001
  - python==3.6.10=h0371630_0
  - libpng==1.6.37=hed695b0_1
  - libiconv==1.15=h516909a_1006
  - libedit==3.1.20181209=hc058e9b_0
  - libgcc-ng==9.1.0=hdf63c60_0
  - jasper==1.900.1=h07fcdf6_1006
  - libllvm9==9.0.1=hc9558a2_0
  - giflib==5.2.1=h516909a_2
  - xorg-libx11==1.6.9=h516909a_0
  - ncurses==6.2=he6710b0_0
  - bzip2==1.0.8=h516909a_2
  - xorg-libxext==1.3.4=h516909a_0
  - xorg-renderproto==0.11.1=h14c3975_1002
  - libwebp==1.0.2=h56121f0_5
  - glib==2.58.3=py36hd3ed26a_1003
  - grpcio==1.27.2=py36hf8bcb03_0
  - xorg-libsm==1.2.3=h84519dc_1000
  - gettext==0.19.8.1=hc5be6a0_1002
  - ffmpeg==4.1.3=h167e202_0
  - lame==3.100=h14c3975_1001
  - x264==1!152.20180806=h14c3975_0
  - libstdcxx-ng==9.1.0=hdf63c60_0
  - cairo==1.16.0=hcf35c78_1003
  - icu==64.2=he1b5a44_1
  - libprotobuf==3.11.4=hd408876_0
  - xorg-libice==1.0.10=h516909a_0
  - libxml2==2.9.10=hee79883_0
  - nspr==4.25=he1b5a44_0
  - nettle==3.4.1=h1bed415_1002
  - scipy==1.4.1=py36habc2bb6_0
  - libxkbcommon==0.10.0=he1b5a44_0
  - pthread-stubs==0.4=h14c3975_1001
  - tensorflow==1.14.0=gpu_py36h3fb9ad6_0
Can you help me with it?

luigifreda, 2020-05-11:
Hi, 
I developed pyslam under Linux. Unfortunately, at the present time, I do not have a mac and can't test the installation process on it. I would love to port pyslam on mac too. 

Nonetheless, I think you can find good suggestions on this [page](https://stackoverflow.com/questions/49154899/resolvepackagenotfound-create-env-using-conda-and-yml-file-on-macos). As far as I understand, the packages you report are not found (hence installed). I would try to test [this](https://stackoverflow.com/questions/49154899/resolvepackagenotfound-create-env-using-conda-and-yml-file-on-macos#50499436). 
1) create a new env configuration `requirements-conda-mac.yml` from [requirements-conda.yml](https://github.com/luigifreda/pyslam/blob/master/requirements-conda.yml) according to this [hint](https://stackoverflow.com/questions/49154899/resolvepackagenotfound-create-env-using-conda-and-yml-file-on-macos#50499436).
2) then modify [pyenv-conda-create.sh](https://github.com/luigifreda/pyslam/blob/master/pyenv-conda-create.sh) in order to call it or simply run 
```
conda env create -f requirements-conda-mac.yml
```
Moreover, I guess you do not have a GPU on your mac, so replace `tensorflow-gpu` with the simple `tensorflow`.

Hope this helps. 
L

luigifreda, 2020-05-11:
You can also try to use this new [requirements file](https://github.com/luigifreda/pyslam/blob/master/requirements-conda-nobuilds.yml) (without build numbers) I've just added . 
```
conda env create -f requirements-conda-nobuilds.yml
```

zhangjun-xyz, 2020-05-12:
I tried. But it also failed. Lots of conflicts are reported:
```
Found conflicts! Looking for incompatible packages.
...
cairo=1.16.0 -> glib[version='>=2.58.3,<3.0a0'] -> libcxx[version='>=4.0.1|>=9.0.0|>=9.0.1']
certifi=2019.11.28 -> python[version='>=2.7,<2.8.0a0'] -> libcxx[version='>=4.0.1|>=9.0.1|>=9.0.0']
expat=2.2.9 -> libcxx[version='>=9.0.1']
pcre=8.44 -> libcxx[version='>=9.0.1']
...
Package ncurses conflicts for:
qt=5.12.5 -> sqlite[version='>=3.30.1,<4.0a0'] -> ncurses[version='>=6.1,<6.2.0a0|>=6.2,<7.0a0']
termcolor=1.1.0 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
wheel=0.34.2 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
absl-py=0.9.0 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
setuptools=46.0.0 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
python_abi=3.6 -> python=3.6 -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
astor=0.8.0 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
tensorflow-estimator=1.14.0 -> python[version='>=3.8,<3.9.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
sqlite=3.31.1 -> libedit[version='>=3.1.20181209,<3.2.0a0'] -> ncurses[version='>=6.1,<7.0a0']
keras-applications=1.0.8 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
gast=0.3.3 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
libedit=3.1.20181209 -> ncurses[version='>=6.1,<7.0a0']
numpy=1.18.1 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
h5py=2.8.0 -> python[version='>=3.7,<3.8.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0']
sqlite=3.31.1 -> ncurses[version='>=6.2,<7.0a0']
nss=3.47 -> sqlite[version='>=3.30.1,<4.0a0'] -> ncurses[version='>=6.1,<6.2.0a0|>=6.2,<7.0a0']
opencv=3.4.1 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
markdown=3.1.1 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
keras-preprocessing=1.1.0 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
scipy=1.4.1 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
werkzeug=1.0.0 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
python=3.6.10 -> ncurses[version='>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
google-pasta=0.1.8 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
readline=7.0 -> ncurses[version='5.9.*|>=6.1,<6.2.0a0|>=6.1,<7.0a0|>=6.0,<7.0a0|6.0.*']
pip=20.0.2 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
tensorflow=1.14.0 -> python=3.7 -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0']
six=1.14.0 -> python -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
ncurses=6.2
protobuf=3.11.4 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
python=3.6.10 -> readline[version='>=7.0,<8.0a0'] -> ncurses[version='5.9.*|>=6.0,<7.0a0|6.0.*']
tensorboard=1.14.0 -> python[version='>=3.8,<3.9.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
glib=2.58.3 -> python[version='>=3.8,<3.9.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']
certifi=2019.11.28 -> python[version='>=2.7,<2.8.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0|>=6.1,<6.2.0a0']
tensorflow-base=1.14.0 -> python[version='>=3.7,<3.8.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0']
wrapt=1.12.1 -> python[version='>=3.6,<3.7.0a0'] -> ncurses[version='>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0']

Package sqlite conflicts for:
astor=0.8.0 -> python[version='>=3.6,<3.7.0a0'] -> sqlite[version='3.13.*|3.20.*|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.28.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.29.0,<4.0a0|>=3.23.1,<4.0a0|>=3.22.0,<4.0a0|>=3.20.1,<4.0a0|>=3.30.0,<4.0a0|>=3.27.2,<4.0a0|>=3.25.1,<4.0a0']
python_abi=3.6 -> python=3.6 -> sqlite[version='3.13.*|3.20.*|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.28.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.29.0,<4.0a0|>=3.23.1,<4.0a0|>=3.22.0,<4.0a0|>=3.20.1,<4.0a0']
scipy=1.4.1 -> python[version='>=3.6,<3.7.0a0'] -> sqlite[version='3.13.*|3.20.*|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.28.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.29.0,<4.0a0|>=3.23.1,<4.0a0|>=3.22.0,<4.0a0|>=3.20.1,<4.0a0|>=3.30.0,<4.0a0|>=3.25.1,<4.0a0|>=3.27.2,<4.0a0']
tensorboard=1.14.0 -> python[version='>=3.8,<3.9.0a0'] -> sqlite[version='3.13.*|3.20.*|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.28.0,<4.0a0|>=3.30.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.29.0,<4.0a0|>=3.27.2,<4.0a0|>=3.23.1,<4.0a0|>=3.22.0,<4.0a0|>=3.20.1,<4.0a0|>=3.25.1,<4.0a0']
[... cut ...]
tensorflow-estimator=1.14.0 -> mock[version='>=2.0.0'] -> funcsigs[version='>=1.0']
tensorflow-base=1.14.0 -> mock[version='>=2.0.0'] -> funcsigs[version='>=1.0']

Package ffmpeg conflicts for:
ffmpeg=4.1.3
opencv=3.4.1 -> ffmpeg[version='>=3.2.3,<3.2.6|>=4.0.1,<4.1.0a0|>=4.0.2,<4.1.0a0']

Package nomkl conflicts for:
libblas=3.8.0 -> openblas[version='>=0.3.6,<0.3.7.0a0'] -> nomkl==3.0=0
blas=1.0 -> openblas -> nomkl==3.0=0
```
I am sorry I am a newbie to anaconda. I know it's tedious so I wanna solve it by myself but I think I'd better reply this message.

zhangjun-xyz, 2020-05-12:
I managed to run the main_vo.py by using the file requirements_pip3.txt.
What I actually did is create an empty conda environment and install the required packages described in requirements_pip3.txt.
I see that there are more requirements in requirements-conda.yml. Does this affect my experiments with pyslam?

luigifreda, 2020-05-12:
I am glad you managed to run main_vo.py.
I would appreciate if you could generate a `requirements-conda-mac.yml` from your conda environment so that we can share it with possible interested people (I will create a section for that in the install instructions, pointing at this issue). You can easily generate such file by running from your new conda environment:
```
$ conda env export > requirements-conda-mac.yml
```
as explained [here](https://github.com/luigifreda/pyslam/blob/master/CONDA.md#user-content-install-packagesenv-from-file). If you succeed to do that, please, let me know also your anaconda/miniconda version and OS version. 

As for your question, tout court, I think the dependencies are managed in a different way. As long as the versions of the main packages are the same (especially TensorFlow stuff), I do not think you'll see big differences. 

Please, let me know if you also succeed to run main_slam.py in your environment. 

L

zhangjun-xyz, 2020-05-16:
Finally I gave up on setting up on Anaconda. I can run main_vo but not main_slam. The g2o built on MacOS is slightly different than that of on Linux. I stopped at this spot. 
Then I turned to Docker. I set up a docker environment of Ubuntu 18.04 on my Mac. It's cool and I finally can run main_vo and main_slam. The result of main_slam are not so good. I will dig out why.
Thanks luigifreda. :)

zhangjun-xyz, 2020-05-16:
Furthermore. To run main_slam.py successfully in my docker environment. I made the following minor changes. 

1. Install pangolin to python packages dir:
[image]
2. Solve UnicodeEncodeError:
    a. initializer.py
[image]
    b. slam.py
[image]
[image]
[image]
[image]

luigifreda, 2020-05-16:
Thank you very much for the update. 

As for main_slam.py, did you succeed to compile cpp stuff (you can check with the script [install_cpp.sh](https://github.com/luigifreda/pyslam/blob/master/install_cpp.sh))?

In any case, try [this parameter change](https://github.com/luigifreda/pyslam/blob/master/TROUBLESHOOTING.md#user-content-bad-tracking-performances), it will slow-down things but performances should be better. 
Please, let me know.

zhangjun-xyz, 2020-05-16:
Yes I've compiled the cpp files successfully. 
And after I changed the parameter kTrackingWaitForLocalMappingToGetIdle to true the tracked points are far more than before. Thanks a lot.
As you said, it became extremely slow. Each frame takes about 20s on my MacBook Pro late 2019.

luigifreda, 2020-05-16:
10s is really a lot! That's the docker-mac mix that makes everything so slow.

closed 2020-06-20 by luigifreda

luigifreda, 2020-07-11:
Hey, now we have an experimental branch `mac`. You can directly install  `pyslam` on your mac (without the need of docker).
Check the instructions [here](https://github.com/luigifreda/pyslam/blob/master/MAC.md)

zhangjun-xyz, 2020-07-12:
Cool, then. 
I am kinda busy these days. I will have a try on this and will let you know how is it going.
Thanks again for your great work.

zhangjun-xyz, 2020-07-20:
@luigifreda I've tested the mac version. It works perfectly, both vo and slam mode. Good job!