luigifreda/pyslam #17: Installation script fails when no opencv-contrib-python was previously installed

michaelschleiss, 2020-05-27:
Installation script fails when opencv-contrib-python is not installed. Following error message occured:

> Cannot uninstall requirement opencv-contrib-python, not installed

michaelschleiss, 2020-05-27:
Also Pangolin will fail to compile if eigen library is not installed.

> /home/parallels/repos/pyslam/thirdparty/pangolin/external/pybind11/include/pybind11/eigen.h:25:10: fatal error: Eigen/Core: No such file or directory

Fixed it with:

> sudo apt-get install libeigen3-dev

michaelschleiss, 2020-05-27:
One more: Had issues with orbslam2_features compilation. 

> CMake Error at CMakeLists.txt:19 (message):
>   OpenCV > 2.4.3 not found.

Fixed it with:

> sudo apt-get install libopencv-dev

luigifreda, 2020-05-27:
Thanks for your help.
I've just updated the scripts according to your suggestions.

closed 2020-05-27 by luigifreda