luigifreda/pyslam #18: opencv problem with conda installation

feritte, 2020-05-29:
Hi
Thanks a lot for this co;plete VO pipeline. I have followed your installation guide 'Install pySLAM in a Custom Python Virtual Environment (conda)' all the requirements installed successfully including opencv-3.4.1 however when I try to run the main_vo.py in the environment, I have the no module found cv2 error. do you have any idea how i can solve this ?

Thanks in advance

luigifreda, 2020-05-29:
Hi,
could you please answer the following points?
- what is your OS? 
- what is your conda setup? Anaconda/Minconda version? 
- could you please check that you have activated the pyslam environment? Check if you have a (pyslam) prefixed prompt on your terminal or run `conda env list` and check that pyslam is shown with a `*` 
- please, run `conda list` within the activated pyslam environment and check that you have opencv in the package list.
- try to run this within your pyslam env 
```
$ python3 -c "import cv2; print(cv2.__version__)"
```

feritte, 2020-05-29:
thanks a lot for your quick answer.

- ubuntu 16.04

- anaconda version : 
    conda version : 4.8.3
    conda-build version : 3.18.11
     python version : 3.7.7.final.0

- pyslam is active (with *)

- conda list in the env ==> "opencv                    3.4.1            py36h6fd60c2_1"

-   python3 -c "import cv2; print(cv2.__version__)" ==> gives me "No module named 'cv2'"

luigifreda, 2020-05-29:
Try this: 
- activate the pyslam env 
- check with `$ which pip` that the considered pip is */pyslam/bin/pip
- run `$ pip install opencv-contrib-python==3.4.2.16`

feritte, 2020-05-29:
thanks a lot for your answer. $ pip install opencv-contrib-python==3.4.2.16 has fixed the problem with opencv. but then it gives me no matplotlib package error (from mplot_thread.py). I have checked the requirements file and I couldnt see the matplotlib package in the file. When I try to install it (i have tried sevral versions), it failed due to the conflicts with many packages installed in the env. At the end I created the environment with  python=3.4.1 from scratch. first I installed matplot and then opencv after I installed all the packages in the list one by one :) finally now I have an environment which let me run the main_vo.py. I dont think that it is the most efficient way to go but at least it works now:) 
thanks a lot again for your guidance and answers

luigifreda, 2020-06-02:
Hi,
thanks for your update. I updated the conda install scripts in order to solve those problems. For the sake of clarity, pip must be used under the wood since conda is not able to manage certain package conflicts. I would appreciate if you could test the new updated scripts (you can pull and follow the same procedure [here](https://github.com/luigifreda/pyslam/blob/master/CONDA.md)).
- Luigi

feritte, 2020-06-03:
Hi Luigi 
I just tested the updated scripts and it worked completely. I can run main_vo without any problem in the created conda environment. I will continue with install_all.sh (by commenting the line "./install_pip3_packages.sh ") in order to run the main_slam.
thanks for the update

ferit

luigifreda, 2020-06-03:
Hi Ferit,
thank you! I am glad to hear that. In order to simplify things, I will add an option to the install scripts in order to jump the installation of pip3 packages on the native python environment. I'll also update the install instructions ;-) 
Best, 
Luigi

feritte, 2020-06-03:
Hi Luigi,

I just finished the complete installation by using the install_all script. When I run the main_slam, it gives me error:
 "self.sift = cv2.xfeatures2d.SIFT_create(AttributeError: module 'cv2.cv2' has no attribute 'xfeatures2d'". 
When I checked the env, i see that there are two different versions of the opencv installed : 
opencv                    3.4.1            py36h6fd60c2_1    anaconda
opencv-contrib-python     3.4.2.16                 pypi_0    pypi
opencv-python             4.2.0.32                 pypi_0    pypi
it might be due to this ? 

thanks,
Ferit

luigifreda, 2020-06-03:
That's weird. You should get the same error with main_vo.py. 
Could you tell me which file that error comes from? (full error log) 
Could you tell me what's now the output of 
```
$ python3 -c "import cv2; print(cv2.__version__)"
```

feritte, 2020-06-03:
error log : 
Traceback (most recent call last):
  File "/home/Workspace/pyslam/main_slam.py", line 76, in <module>
    feature_tracker = feature_tracker_factory(**tracker_config)
  File "/home/Workspace/pyslam/feature_tracker.py", line 66, in feature_tracker_factory
    tracker_type = tracker_type)
  File "/home/Workspace/pyslam/feature_tracker.py", line 201, in __init__
    descriptor_type=descriptor_type)                     
  File "/home/Workspace/pyslam/feature_manager.py", line 97, in feature_manager_factory
    return FeatureManager(num_features, num_levels, scale_factor, detector_type, descriptor_type)
  File "/home/Workspace/pyslam/feature_manager.py", line 425, in __init__
    self._feature_detector = ContextDescFeature2D(num_features=self.num_features)
  File "/home/Workspace/pyslam/feature_contextdesc.py", line 115, in __init__
    'upright': False})       
  File "/home/Workspace/pyslam/thirdparty/contextdesc/models/base_model.py", line 61, in __init__
    self._init_model()
  File "/home/Workspace/pyslam/thirdparty/contextdesc/models/loc_model.py", line 37, in _init_model
    self.sift_wrapper.create()
  File "/home/Workspace/pyslam/thirdparty/contextdesc/utils/opencvhelper.py", line 47, in create
    self.sift = cv2.xfeatures2d.SIFT_create(
AttributeError: module 'cv2.cv2' has no attribute 'xfeatures2d'
Exception ignored in: <bound method ContextDescFeature2D.__del__ of <feature_contextdesc.ContextDescFeature2D object at 0x7f009e463940>>
Traceback (most recent call last):
  File "/home/Workspace/pyslam/feature_contextdesc.py", line 123, in __del__
    self.loc_model.close()
AttributeError: 'ContextDescFeature2D' object has no attribute 'loc_model'

**** 
python3 -c "import cv2; print(cv2.__version__)" ==> 4.2.0

luigifreda, 2020-06-03:
Try this HACK: reinstall opencv-contrib-python by running 
```
$ pip install opencv-contrib-python==3.4.2.16
```

In any case, I have created and pushed an `install_all_conda.sh` script. Please, remove the old `pyslam` environment by running `$ . pyenv-conda-delete.sh` and follow the new instructions [here](https://github.com/luigifreda/pyslam/blob/master/CONDA.md#user-content-full-conda-installation). 

I am not sure I will be able to reproduce your error since I'm on ubuntu 18.

feritte, 2020-06-03:
yes definitely it worked after reinstalling opencv hack.
Thanks a lot, Luigi.

luigifreda, 2020-06-03:
Great! Glad to hear that. We can close this issue :-)

closed 2020-06-03 by luigifreda