luigifreda/pyslam #20: VO with features in a plane

michaelschleiss, 2020-06-04:
Is it possible to use the VO Pipeline with features that all lie in the same plane e.g. capturing the ground in a downward looking camera?

luigifreda, 2020-06-04:
Yes, you can use main_slam.py. 
In general, with an RBG camera, there can be issues if you look at textureless portions of the ground or if there are self-similarities (e.g. tiles) or perceptual aliasing. 
As for tracking features that lie on a plane, take a look at the note above the function estimate_pose_ess_mat() [here](https://github.com/luigifreda/pyslam/blob/master/utils_geom.py#L234). This function is used by the initializer.

michaelschleiss, 2020-06-04:
Thanks for your comment! The main_slam.py pipeline does indeed work very well for my dataset. Whereas main_vo.py's pipeline doesn't. I would like to understand the issue a little more. 

I had assumed that this might be caused by the features being all in one plane. However, the notes from your link mention that the 5-point algorithm used in the OpenCV implementation should work well in such cases. 

I created my dataset by simulating a drone flying a straight line over a city based on Google-Maps-like satellite imagery. It depicts buildings and street segments. Therefore, self-similarities or textureless portions should not be an issue. Do you have an idea why the main_vo.py pipeline creates such a curvy trajectory?

Here is a visualization of the trajectory created by main_slam.py

[image]

Here is the one from main_vo.py

[image]

luigifreda, 2020-06-05:
I would consider `main_vo.py` as a toy/educational example (as explained in the README). It actually needs the knowledge of a groundtruth for estimating the inter-frame scale. 
First of all, did you provide `main_vo.py` a reasonable groundtruth for inter-frame scale estimation? 

Moreover, `main_vo.py` uses `estimate_pose_ess_mat()` at every single frame and only take into account feature/data matching/association between _pairs_ of consecutive frames.
As explained in the [linked notes](https://github.com/luigifreda/pyslam/blob/master/utils_geom.py#L223) there are (i) degenerate arrangements of features and (ii) degenerate motions (e.g pure rotations, very small motions, etc.) for `estimate_pose_ess_mat()`. So features lying on a plane may be just a part of the problems. 
Does your drone stop or perform pure rotations or move very slowly? Its motions may be close to degenerate and hence problematic for `estimate_pose_ess_mat()`.

`main_slam.py` uses `estimate_pose_ess_mat()` only for initializing the map. Moreover, `main_slam.py` adds feature tracking (data association) along _multiple_ frames, point triangulation, keyframe management, and bundle adjustment in order to better and more robustly estimate the camera trajectory.

I just hinted at the main reasons for the different performances... but there is even more ;-)

luigifreda, 2020-06-06:
I hope those lines gave you an idea. I am closing this.

closed 2020-06-06 by luigifreda

michaelschleiss, 2020-06-10:
> Does your drone stop or perform pure rotations or move very slowly? Its motions may be close to degenerate and hence problematic for `estimate_pose_ess_mat()`.

No pure rotations or stops. The drone flies a straight line with continuous speed. I also tried to increase the inter-frame displacement. I guess the scenario with a downloading camera observing a flat scene is not suitable for the 5-point algorithm. I have to study the problem a little more. I think your hints will be of great help. Thank you very much!