luigifreda/pyslam #23: trajectory in a loop

feritte, 2020-06-05:
I am testing main_slam.py with a short video that i captured in a parking with a dashcam. It contains one big loop at low speed. I observe big drift and scaling issue especially at the second half. Moreover at the beginning and at each turn (90 degree), I see multiple path in the map viewer. Do you think that it is normal or maybe i do something wrong with  the settings ? Here is i am adding two screenshot which shows the trajectory and the map
[image]

[image]

Thanks

luigifreda, 2020-06-06:
I would start considering the following points: 
- Did you correctly calibrate your camera? 
- If you use the camera behind a windshield, the windshield itself may act as an additional 'distorting' lens, therefore, you need further calibration for that. 
- Does your camera have autofocus? If yes, that can change your calibration parameters and continuously break the assumption of a constant calibration matrix. 

In general, in order to have a better understanding of what's going on, one needs to take a look at the video and consider all the parameter settings (if you changed them), the used features, etc.

feritte, 2020-06-06:
Thanks for your answer. I'll definetely verify calibration and redo if it is necessary considering your remarks. What is the color code (Blue vs green path) in the mapping window?

luigifreda, 2020-06-06:
- [covisibility graph in green](https://github.com/luigifreda/pyslam/blob/master/viewer3D.py#L217)
- [spanning tree in blue](https://github.com/luigifreda/pyslam/blob/master/viewer3D.py#L223) 
take a look at [ORBSLAM paper](http://webdiis.unizar.es/~raulmur/MurMontielTardosTRO15.pdf)

feritte, 2020-06-08:
thanks a lot for your answer. I am reading the paper for refreshing the theory. Is it also possible that the warnings that I am receiving at the beginning of the main_slam.py can cause this performance issue ? 
 

-WARNING: cannot import ORBextractor from orbslam2_features, check the file TROUBLESHOOTING.md

-WARNING: cannot import pyslam_utils

-WARNING: cannot import Orbslam2Feature2D from feature_orbslam2, check the file TROUBLESHOOTING.md

luigifreda, 2020-06-08:
Those warnings mean that you didn't correctly compile C++ libraries (wrapped with pybind11) or your working python environment does not succeed to correctly import them. You may want to test the new `install_all_conda.sh` script. 
I don't know which local features you are actually using. Consider that there are local features in the list that are not really working well as expected.

feritte, 2020-06-08:
well I run the new script and all the installation thirdparty included was successful but I have     import g2o
ModuleNotFoundError: No module named 'g2o' error when I tried to run main_slam.py

feritte, 2020-06-08:
deleting build ../g2o/build and repeating the g2o installation fixed that error.. however the same warnings are still there.

luigifreda, 2020-06-09:
You can use the script `clean.sh` in order to clean things in a proper way. Then, you have to delete your pyslam environment with `pyenv-conda-delete.sh` and launch `install_all_conda.sh` again.

feritte, 2020-06-09:
i followed the steps. but still i have the same warnings therefore when I choose orb2 feature FeatureTrackerConfigs.ORB2 ==> self._feature_detector = Orbslam2Feature2D(self.num_features, self.scale_factor, orb2_num_levels) 
TypeError: 'NoneType' object is not callable

luigifreda, 2020-06-09:
Worked again on that. I've just updated the install scripts. You can now retry and everything should work.

feritte, 2020-06-09:
thanks a lot Luigi. latest update has fixed the problems with binding. I dont have warnings anymore...:) 
i will try my small video sequence again with my new calibration file. Just to be sure, for example videos from kittim there is always groundtruth.txt file and even main_slam.py also reads it but it needs this file in order to have good results ? 
thanks again

luigifreda, 2020-06-10:
I'm glad it worked. 
For the sake of clarity, I have set `groundtruth=None` in main_slam.py. The ground-truth is not used at all by the Slam() class. You may want to better analyze the code and study the suggested references before actually keeping on with further experiments :-)

closed 2020-06-10 by luigifreda