luigifreda/pyslam #24: pybind problem

nidetaoge, 2020-06-09:
Nice work! However I have met some prblems and thanks for helping!

Main problem:
python -O main_vo.py 
 WARNING: cannot import ORBextractor from orbslam2_features, check the file TROUBLESHOOTING.md
 WARNING: cannot import pyslam_utils
 WARNING: cannot import Orbslam2Feature2D from feature_orbslam2, check the file TROUBLESHOOTING.md
 WARNING: cannot import KeyNetDescFeature2D from feature_keynet, check the file TROUBLESHOOTING.md

I use conda to create pyslam env, anything is fine except that I can not use orbslam2_features.
In pyslam env, I use python3.6, but in thildparty/orbslam2_features/CmakLists,  the result of find_package(PythonLibs) is the one in system(python3.5), and the lib is 
orbslam2_features.cpython-35m-x86_64-linux-gnu.so.
Even I SET(PYTHON_INCLUDE_DIRS anaconda3/envs/pyslam/include/python3.6m), or set PYTHONPATH, the lib is 
orbslam2_features.cpython-35m-x86_64-linux-gnu.so(not the 36m one), and when I do python -O main_vo.py, I still face the problem that I can not use ORBextractor. Can you help me to solve this?

luigifreda, 2020-06-09:
Hi, let's try [this](https://github.com/luigifreda/pyslam/blob/master/TROUBLESHOOTING.md#user-content-cannot-properly-import-g2o-library-or-other-libs): 
- go in the root folder of pyslam 
- run `$ clean.sh`
- launch again `$ . install_all_conda.sh  `  (N.B.: do not forget the dot! (without '/' !))

luigifreda, 2020-06-09:
Worked again on that. I've just updated the install scripts. If you tried the procedure above and it didn't work, you could retry now and everything should work.

closed 2020-06-11 by luigifreda