luigifreda/pyslam #25: CMake error while compiling g2opy OPENGL_gl_LIBRARY (ADVANCED)

nikkkkhil, 2020-06-19:
I'm trying to install pyslam2 in my system. 
System Information :
ubuntu 18.04 
architecture - x86_64
conda environment 
I installed all basic libraries by ./install_basic.sh then when I try to install  libs pangolin, g2opy I get CMake Error as follow 
```
[Installing delf ...                                                                                                                                           [118/1615]
================================================ 
Building and installing cpp ... 
-- found OpenCV version: 4.3.0
-- opencv include: /usr/local/include/opencv4
-- opencv lib dirs: /usr/local
-- opencv libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_gapi;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdete
ct;opencv_photo;opencv_stitching;opencv_video;opencv_videoio;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dnn_objdetect;opencv_dnn
_superres;opencv_dpm;opencv_face;opencv_freetype;opencv_fuzzy;opencv_hdf;opencv_hfs;opencv_img_hash;opencv_intensity_transform;opencv_line_descriptor;opencv_optflow;ope
ncv_phase_unwrapping;opencv_plot;opencv_quality;opencv_rapid;opencv_reg;opencv_rgbd;opencv_saliency;opencv_shape;opencv_stereo;opencv_structured_light;opencv_superres;o
pencv_surface_matching;opencv_text;opencv_tracking;opencv_videostab;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
-- pybind11 v2.3.dev0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nikhil/Desktop/practice/pyslam/cpp/utils/build
[100%] Built target pyslam_utils
================================================ 
Building Thirdparty 
================================================ 
================================================
Configuring and building thirdparty/Pangolin ... 
================================================================== 
Configuring and building thirdparty/g2o ... 
-- Compiling on Unix
-- Found CHOLMOD and its dependencies
-- Building LGPL code as static library (affects license of the binary)
-- Compiling with OpenGL support
-- Could NOT find QGLVIEWER (missing: QGLVIEWER_INCLUDE_DIR QGLVIEWER_LIBRARY)
-- Compiling with GCC
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
OPENGL_gl_LIBRARY (ADVANCED)
    linked by target "freeglut_minimal" in directory /home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/EXTERNAL/freeglut
    linked by target "types_slam2d" in directory /home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/g2o/types/slam2d
    linked by target "types_slam3d" in directory /home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/g2o/types/slam3d
    linked by target "types_sclam2d" in directory /home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/g2o/types/sclam2d
    linked by target "types_slam2d_addons" in directory /home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/g2o/types/slam2d_addons
    linked by target "types_slam3d_addons" in directory /home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/g2o/types/slam3d_addons

-- Configuring incomplete, errors occurred!
See also "/home/nikhil/Desktop/practice/pyslam/thirdparty/g2opy/build/CMakeFiles/CMakeOutput.log".](url)
](url)
```
How can I fix this ?

luigifreda, 2020-06-19:
Try this 
sudo apt-get install -y libqglviewer-dev

luigifreda, 2020-06-19:
If you want to install pyslam under conda, please, take a look a the README and refer to this page https://github.com/luigifreda/pyslam/blob/master/CONDA.md as explained

closed 2020-06-20 by luigifreda

nikkkkhil, 2020-06-20:
Thanks @luigifreda . I was able to fix issue following [this](https://github.com/colmap/colmap/issues/713#issuecomment-558505243l)