luigifreda/pyslam #28: Building in pyslam utils

shumshersubashgautam, 2020-07-11:
Plzz help me to solve this problem.
Scanning dependencies of target pyslam_utils
[ 33%] Building CXX object CMakeFiles/pyslam_utils.dir/utils_module.cpp.o
[ 66%] Building CXX object CMakeFiles/pyslam_utils.dir/utils.cpp.o
In file included from /home/intelligence/pyslam/pyslam/cpp/utils/utils.cpp:2:0:
/home/intelligence/pyslam/pyslam/cpp/utils/utils.h:20:94: error: ‘WARP_FILL_OUTLIERS’ is not a member of ‘cv’
                           const int warp_flags= cv::WARP_INVERSE_MAP + cv::INTER_CUBIC + cv::WARP_FILL_OUTLIERS);
                                                                                              ^~~~~~~~~~~~~~~~~~
/home/intelligence/pyslam/pyslam/cpp/utils/utils.h:27:94: error: ‘WARP_FILL_OUTLIERS’ is not a member of ‘cv’
                           const int warp_flags= cv::WARP_INVERSE_MAP + cv::INTER_CUBIC + cv::WARP_FILL_OUTLIERS);
                                                                                              ^~~~~~~~~~~~~~~~~~
CMakeFiles/pyslam_utils.dir/build.make:86: recipe for target 'CMakeFiles/pyslam_utils.dir/utils.cpp.o' failed
make[2]: *** [CMakeFiles/pyslam_utils.dir/utils.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/intelligence/pyslam/pyslam/cpp/utils/utils_module.cpp:9:0:
/home/intelligence/pyslam/pyslam/cpp/utils/utils.h:20:94: error: ‘WARP_FILL_OUTLIERS’ is not a member of ‘cv’
                           const int warp_flags= cv::WARP_INVERSE_MAP + cv::INTER_CUBIC + cv::WARP_FILL_OUTLIERS);
                                                                                              ^~~~~~~~~~~~~~~~~~
/home/intelligence/pyslam/pyslam/cpp/utils/utils.h:27:94: error: ‘WARP_FILL_OUTLIERS’ is not a member of ‘cv’
                           const int warp_flags= cv::WARP_INVERSE_MAP + cv::INTER_CUBIC + cv::WARP_FILL_OUTLIERS);
                                                                                              ^~~~~~~~~~~~~~~~~~
/home/intelligence/pyslam/pyslam/cpp/utils/utils_module.cpp: In function ‘void pybind11_init_pyslam_utils(pybind11::module&)’:
/home/intelligence/pyslam/pyslam/cpp/utils/utils_module.cpp:37:152: error: ‘WARP_FILL_OUTLIERS’ is not a member of ‘cv’
      "image"_a,"kps"_a,"patch_size"_a,"use_orientation"_a=true,"scale_factor"_a=1.0,"warp_flags"_a=cv::WARP_INVERSE_MAP + cv::INTER_CUBIC + cv::WARP_FILL_OUTLIERS
                                                                                                                                                 ^~~~~~~~~~~~~~~~~~
CMakeFiles/pyslam_utils.dir/build.make:62: recipe for target 'CMakeFiles/pyslam_utils.dir/utils_module.cpp.o' failed
make[2]: *** [CMakeFiles/pyslam_utils.dir/utils_module.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pyslam_utils.dir/all' failed
make[1]: *** [CMakeFiles/pyslam_utils.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

luigifreda, 2020-07-12:
Could you please specify your OS? 
Did you use one of the provided install scripts? All the install scripts install a proper OpenCV version.
It seems that your configuration includes the installation of a very old OpenCV version.
Even OpenCV 3.0.0 include the definition of  cv::WARP_FILL_OUTLIERS 
https://docs.opencv.org/3.0.0/da/d54/group__imgproc__transform.html

shumshersubashgautam, 2020-07-15:
I tested those in ubuntu  18.04 errors is due to open cv right versions errors  i solved the errors  main_vo.py but  main_slam.py during building pangolin to original  orbslam2 repository  i got lots of errors of eigen3 to..g2opy. how i can solve those problems.now i turned to ubuntu 16.04

shumshersubashgautam, 2020-07-15:
@luigifreda with these errors.. and plz if u have any ideas to solve the problems of orbslam2 eigens versions to other dependencies errors  plz suggest me.. and from ur repo i got this errors in building ur install_all.sh 
Scanning dependencies of target _pangolin
[  1%] Building CXX object src/CMakeFiles/_pangolin.dir/utils/file_extension.cpp.o
[  2%] Building CXX object src/CMakeFiles/_pangolin.dir/utils/file_utils.cpp.o
[  3%] Building CXX object src/CMakeFiles/_pangolin.dir/utils/threadedfilebuf.cpp.o
[  4%] Building CXX object src/CMakeFiles/_pangolin.dir/utils/sigstate.cpp.o
[  5%] Building CXX object src/CMakeFiles/_pangolin.dir/image/image_io.cpp.o
[  6%] Building CXX object src/CMakeFiles/_pangolin.dir/utils/uri.cpp.o
[  7%] Building CXX object src/CMakeFiles/_pangolin.dir/utils/timer.cpp.o
[  8%] Building CXX object src/CMakeFiles/_pangolin.dir/image/image_io_exr.cpp.o
[  9%] Building CXX object src/CMakeFiles/_pangolin.dir/image/image_io_jpg.cpp.o
In file included from /usr/include/OpenEXR/ImfHeader.h:51:0,
                 from /usr/include/OpenEXR/ImfInputFile.h:46,
                 from /content/pyslam/thirdparty/pangolin/src/image/image_io_exr.cpp:8:
/usr/include/OpenEXR/ImathVec.h:228:34: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     const Vec2 & normalizeExc () throw (IEX_NAMESPACE::MathExc);
                                  ^~~~~
/usr/include/OpenEXR/ImathVec.h:232:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     Vec2<T>  normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                     ^~~~~
/usr/include/OpenEXR/ImathVec.h:440:34: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     const Vec3 & normalizeExc () throw (IEX_NAMESPACE::MathExc);
                                  ^~~~~
/usr/include/OpenEXR/ImathVec.h:444:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     Vec3<T>  normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                     ^~~~~
/usr/include/OpenEXR/ImathVec.h:622:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     const Vec4 &    normalizeExc () throw (IEX_NAMESPACE::MathExc);
                                     ^~~~~
/usr/include/OpenEXR/ImathVec.h:626:44: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     Vec4<T>         normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                            ^~~~~
/usr/include/OpenEXR/ImathVec.h:714:30: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec2<short>::normalizeExc () throw (IEX_NAMESPACE::MathExc);
                              ^~~~~
/usr/include/OpenEXR/ImathVec.h:723:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec2<short>::normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                     ^~~~~
/usr/include/OpenEXR/ImathVec.h:738:28: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec2<int>::normalizeExc () throw (IEX_NAMESPACE::MathExc);
                            ^~~~~
/usr/include/OpenEXR/ImathVec.h:747:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec2<int>::normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                   ^~~~~
/usr/include/OpenEXR/ImathVec.h:762:30: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec3<short>::normalizeExc () throw (IEX_NAMESPACE::MathExc);
                              ^~~~~
/usr/include/OpenEXR/ImathVec.h:771:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec3<short>::normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                     ^~~~~
/usr/include/OpenEXR/ImathVec.h:786:28: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec3<int>::normalizeExc () throw (IEX_NAMESPACE::MathExc);
                            ^~~~~
/usr/include/OpenEXR/ImathVec.h:795:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec3<int>::normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                   ^~~~~
/usr/include/OpenEXR/ImathVec.h:809:30: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec4<short>::normalizeExc () throw (IEX_NAMESPACE::MathExc);
                              ^~~~~
/usr/include/OpenEXR/ImathVec.h:818:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec4<short>::normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                     ^~~~~
/usr/include/OpenEXR/ImathVec.h:833:28: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec4<int>::normalizeExc () throw (IEX_NAMESPACE::MathExc);
                            ^~~~~
/usr/include/OpenEXR/ImathVec.h:842:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec4<int>::normalizedExc () const throw (IEX_NAMESPACE::MathExc);
                                   ^~~~~
/usr/include/OpenEXR/ImathVec.h:1212:26: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 Vec2<T>::normalizeExc () throw (IEX_NAMESPACE::MathExc)
[... 155 lines cut ...]
[... cut ...]
examples/SimpleDisplay/CMakeFiles/SimpleDisplay.dir/build.make:102: recipe for target 'examples/SimpleDisplay/SimpleDisplay' failed
make[2]: *** [examples/SimpleDisplay/SimpleDisplay] Error 1
CMakeFiles/Makefile2:603: recipe for target 'examples/SimpleDisplay/CMakeFiles/SimpleDisplay.dir/all' failed
make[1]: *** [examples/SimpleDisplay/CMakeFiles/SimpleDisplay.dir/all] Error 2
[ 87%] Linking CXX shared module ../../pangolin.cpython-36m-x86_64-linux-gnu.so
[ 87%] Built target pangolin
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

luigifreda, 2020-08-01:
Given this: 
`/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libGL.so: undefined reference to _glapi_tls_Current' collect2: error: ld returned 1 exit status examples/HelloPangolin/CMakeFiles/HelloPangolin.dir/build.make:102: recipe for target 'examples/HelloPangolin/HelloPangolin' failed make[2]: *** [examples/HelloPangolin/HelloPangolin] Error 1 CMakeFiles/Makefile2:328: recipe for target 'examples/HelloPangolin/CMakeFiles/HelloPangolin.dir/all' failed make[1]: *** [examples/HelloPangolin/CMakeFiles/HelloPangolin.dir/all] Error 2 make[1]: *** Waiting for unfinished jobs.... /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libGL.so: undefined reference to _glapi_tls_Current'`
you need to install opengl on your system 
https://medium.com/@theorose49/install-opengl-at-ubuntu-18-04-lts-31f368d0b14e

closed 2020-08-01 by luigifreda