luigifreda/pyslam #30: Videos without groundtruth

adizhol, 2020-07-20:
Hello,

Thank you for this useful repo.
How can I run the vo and slam on videos without ground truth?

Thanks

mihailk239, 2020-07-20:
You can generate zero groundtruth

luigifreda, 2020-08-01:
Hi @adizhol 
thank you for the message. 
As explained in the README, you need a ground truth only with` main_vo.py`. You do **NOT** need any grountruth with `main_slam.py`. 
So, when running `main_slam.py` you can actually ignore the groundtruth entry in `config.ini`: the script will actually ignore it as well 
https://github.com/luigifreda/pyslam/blob/master/main_slam.py#L58

closed 2020-08-01 by luigifreda

abcabc456, 2024-01-31:
Hello，
Thank you for this useful repo.
What is the format of groundtruth for various data types？
Thanks