luigifreda/pyslam #33: Can't find 'install_all_mac_venv.sh' file

Brian-ZhenLiu, 2020-08-10:
To install pyslam under macOS, I just followed the instructions to run './install_all_mac_venv.sh', however, there is not a file's name is 'install_all_mac_venv.sh'.  So, I just wonder that how to launch the macOS install script? Thanks.

luigifreda, 2020-08-13:
Hi,
thanks for your interest. You need to read the instructions [here](https://github.com/luigifreda/pyslam/blob/master/MAC.md).

Please, follow these install steps:
1. first you need to install homebrew and Xcode CLT 
2. download this repo and **move to the experimental mac branch**:
```
   $ git clone https://github.com/luigifreda/pyslam.git 
   $ cd pyslam 
   $ git checkout mac 

```
Did you run `$ git checkout mac `? Once you've run it, you will find the file `install_all_mac_venv.sh`

closed 2020-08-13 by luigifreda

Brian-ZhenLiu, 2020-08-17:
Thanks for your reply. It works.
When I run the VO version, there is an error occurs: 'ERROR while reading the file' when it tried to read the video in the end. May I ask how to deal with this?
 [image]

luigifreda, 2020-08-26:
Can you play that file separately?