luigifreda/pyslam #36: Installing all in python virtual env on ubuntu 20.04

kevkid, 2020-10-16:
If you get this error using opencv 4.x as found [here](https://answers.opencv.org/question/225224/opencvcvh-not-found/):
```
pyslam/thirdparty/orbslam2_features/ORBextractor.h:26:10: fatal error: opencv/cv.h: No such file or directory
   26 | #include <opencv/cv.h>
```
just go to the file ORBextractor.h and comment out the include using //#include <opencv.h>

If you run into this error:
Building CXX object python/CMakeFiles/g2o.dir/g2o.cpp.o
........
make[2]: *** [python/CMakeFiles/g2o.dir/build.make:63：python/CMakeFiles/g2o.dir/g2o.cpp.o] error 1
make[1]: *** [CMakeFiles/Makefile2:1345：python/CMakeFiles/g2o.dir/all] error 2
make: *** [Makefile:130：all] error 2

The fix is found [here](https://github.com/uoip/g2opy/issues/46) by doing this:
```
        .def("x", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::x)
        .def("y", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::y)
        .def("z", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::z)
        .def("w", (double (Eigen::Quaterniond::*) () const) &Eigen::Quaterniond::w)
```
in g2opy/python/core/eigen_types.h to:

```
        .def("x", [](const Eigen::Quaterniond& q) { return q.x(); })
        .def("y", [](const Eigen::Quaterniond& q) { return q.y(); })
        .def("z", [](const Eigen::Quaterniond& q) { return q.z(); })
        .def("w", [](const Eigen::Quaterniond& q) { return q.w(); })
```

Also you may have issues installing opencv-contrib so you also need to install some stuff via pip manually:
`pip3 install scipy opencv-contrib-python termcolor ordered_set matplotlib pygame pyopengl `

This got it to build and run except for this error:

```
(pyslam) ~/pyslam$ python3 -O main_slam.py
 WARNING: cannot import SuperPointFeature2D from feature_superpoint, check the file TROUBLESHOOTING.md
 WARNING: cannot import TfeatFeature2D from feature_tfeat, check the file TROUBLESHOOTING.md
 WARNING: cannot import HardnetFeature2D from feature_hardnet, check the file TROUBLESHOOTING.md
 WARNING: cannot import GeodescFeature2D from feature_geodesc, check the file TROUBLESHOOTING.md
 WARNING: cannot import SosnetFeature2D from feature_sosnet, check the file TROUBLESHOOTING.md
 WARNING: cannot import L2NetFeature2D from feature_l2net, check the file TROUBLESHOOTING.md
 WARNING: cannot import LogpolarFeature2D from feature_logpolar, check the file TROUBLESHOOTING.md
 WARNING: cannot import D2NetFeature2D from feature_d2net, check the file TROUBLESHOOTING.md
 WARNING: cannot import DelfFeature2D from feature_delf, check the file TROUBLESHOOTING.md
 WARNING: cannot import ContextDescFeature2D from feature_contextdesc, check the file TROUBLESHOOTING.md
 WARNING: cannot import LfNetFeature2D from feature_lfnet, check the file TROUBLESHOOTING.md
 WARNING: cannot import R2d2Feature2D from feature_r2d2, check the file TROUBLESHOOTING.md
 WARNING: cannot import KeyNetDescFeature2D from feature_keynet, check the file TROUBLESHOOTING.md
importing pygame
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
initialising pygame
Processing Video Input
num frames:  1206
fps:  29.836205991321275
tracker_config:  {'num_features': 2000, 'num_levels': 8, 'scale_factor': 1.2, 'detector_type': <FeatureDetectorTypes.ORB2: 7>, 'descriptor_type': <FeatureDescriptorTypes.ORB2: 5>, 'match_ratio_test': 0.7, 'tracker_type': <FeatureTrackerTypes.DES_BF: 1>}
num_levels:  8
using opencv  4.4.0
Using Orbslam2Feature2D
 using same detector and descriptor object:  ORB2
num_levels:  8
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/pyslam/viewer3D.py", line 84, in viewer_thread
    self.viewer_init(kViewportWidth, kViewportHeight)
  File "/home/user/pyslam/viewer3D.py", line 92, in viewer_init
    pangolin.CreateWindowAndBind('Map Viewer', w, h)
AttributeError: module 'pangolin' has no attribute 'CreateWindowAndBind'
```

luigifreda, 2020-11-08:
I added an install procedure that is specific for Ubuntu 20.04. Check [here](https://github.com/luigifreda/pyslam#install-pyslam-under-ubuntu-2004). Let me know if it works for you.   

In particular, move into the experimental branch `ubuntu20`
```
$ git checkout ubuntu20  
```
and then follow the instructions for creating a new virtual environment `pyslam` described [here](https://github.com/luigifreda/pyslam/blob/master/PYTHON-VIRTUAL-ENVS.md).

closed 2020-11-08 by luigifreda

alexriedel1, 2021-02-17:
Anyone else having problem building pangolin, this helped me: 

https://github.com/stevenlovegrove/Pangolin/issues/27#issuecomment-281009363