luigifreda/pyslam #41: How can I run main_slam.py to generate estimated trajectory file?

linpeisensh, 2020-12-16:
Thank you for your excellent work!

I have run main_slam.py with tum dataset and want to generate trajectory file as tum given groundtruth.txt.

Is the slam.tracking.tracking_history.relative_frame_poses the history of estimated trajectory of slam?
If yes, I have tried to print the last element of  slam.tracking.tracking_history.relative_frame_poses and its timestamp, but it's different from the groundtruth.
In ground truth:
1341845986.0403 -0.2093 -2.9924 1.3790 -0.7683 0.0502 -0.1037 0.6296
my result:
1341845986.0421 0.02307 0.0387 0.0204 -0.0093 0.0049, -0.0108  0.9998

I use this code to print my result:
```
from scipy.spatial.transform import Rotation as R
print(timestamp)
print(R.from_matrix(slam.tracking.tracking_history.relative_frame_poses[-1].orientation().matrix()))
print(slam.tracking.tracking_history.relative_frame_poses[-1].position())
```

Do I misunderstand something?
Do I need to calibrate camera for tum dataset?
Could you give me some advice to generate this estimated trajectory file?

linpeisensh, 2020-12-17:
It's weird for me.
The ATE of slam is bigger than vo.
[image]
Here is my definition of trajectory of slam:

```
trajectory = []
n_his = len(st_his.relative_frame_poses)
for i in range(n_his):
	if st_his.slam_states[i] == SlamState.OK: #OK
		cur_pose = CameraPose(st_his.relative_frame_poses[i])
		cur_tra = [str(round(st_his.timestamps[i], 4))] + list(map(str, np.round(cur_pose.Ow, decimals=4))) + \
				  list(map(str, np.round(R.from_matrix(cur_pose.Rcw).as_quat(), decimals=4)))
		trajectory.append(cur_tra)
```

Do I misunderstand something?
@luigifreda 
Thank you in advance!

luigifreda, 2020-12-19:
Hi, 

1) you cannot compare main_slam.py with main_vo.py. 
Did you study the two methods? As explained in the README

`
main_vo.py combines the simplest VO ingredients without performing any image point triangulation or windowed bundle adjustment. At each step $k$, main_vo.py estimates the current camera pose $C_k$ with respect to the previous one $C_{k-1}$. The inter-frame pose estimation returns $[R_{k-1,k},t_{k-1,k}]$ with $||t_{k-1,k}||=1$. With this very basic approach, you need to use a **ground truth** in order to recover a correct inter-frame scale $s$ and estimate a valid trajectory by composing $C_k = C_{k-1} * [R_{k-1,k}, s t_{k-1,k}]$. This script is a first start to understand the basics of inter-frame feature tracking and camera pose estimation.`

Therefore, main_vo.py is using the ground truth in order to retrieve the interframe scale! 
On the other hand, main_slam.py is estimating the trajectory up-to-scale given it's a pure monocular approach, without using any information coming from the ground truth. 

2) you cannot compare the slam/vo trajectories directly with the ground truth. 
In order to compare trajectories, you need to align them by using a package like 
https://github.com/MichaelGrupp/evo 
and considering the manifolds ( SE(3) or Sim(3) ) in which the methods compute their estimates.

3) in slam.tracking.tracking_history there are keyframes references and relative frame poses. 
There is not a function that dumps the final computed trajectory at the end of the run (after all the bundle adjustments).  But it easy to code it if you need it.

closed 2020-12-19 by luigifreda

linpeisensh, 2020-12-19:
Thank you for your reply!
Here is my estimation trajectory code after slam process is done :
```
trajectory = []
n_his = len(st_his.relative_frame_poses)
for i in range(n_his):
	if st_his.slam_states[i] == SlamState.OK: #OK
		cur_pose = CameraPose(st_his.relative_frame_poses[i])
		cur_tra = [str(round(st_his.timestamps[i], 4))] + list(map(str, np.round(cur_pose.Ow, decimals=4))) + \
				  list(map(str, np.round(R.from_matrix(cur_pose.Rcw).as_quat(), decimals=4)))
		trajectory.append(cur_tra)
```
Could you help me to check, whether it is correct? Or could you show me what could you generate this trajectory file?