luigifreda/pyslam #42: What is the setting for "main-slam-kitti-map"?

hyeonhoshin, 2020-12-18:
Dear luigifreda,

I'm trying to find in order to find the parameter settings same as "main-slam-kitti-map".

But, even though I tried increasing  _kTrackingWaitForLocalMappingSleepTime_, and _kTrackingWaitForLocalMappingToGetIdle=True_, a path diverges at 800 frame in kitti 06 seq.

Could you please let me know **the parameters used for generating "main-slam-kitti-map"** using this program, please?

(I guessed you used SUPERPOINT and some parameter tweak.

But I failed all possible combination I knew)

Thank you.

[image]

luigifreda, 2020-12-19:
I've just taken this snapshot this morning without touching anything (default config with ORB2)

[image]

closed 2020-12-19 by luigifreda

hyeonhoshin, 2020-12-19:
Thank you so much!