luigifreda/pyslam #43: AttributeError: module 'g2o' has no attribute 'Flag'

JeffreyYH, 2021-03-14:
Hi, when I tried to run main_slam.py, it shows AttributeError: module 'g2o' has no attribute 'Flag'.

Thank you!

luigifreda, 2021-03-21:
Hi,
some questions:
- which is your OS? 
- which install script did you use? 

That message means pyslam is not using the custom g2o that needs to be compiled via the install scripts.

JeffreyYH, 2021-03-21:
Thanks for your reply Luigi,
 I am using Ubuntu 18.04 and I used install_thirdparty.sh to install g2o.

luigifreda, 2021-03-21:
I cleaned with `$ ./clean.sh` and launched again a new install with `./install_all.sh` and everything runs correctly. 
Can you please check that you have a file like `thirdparty/g2opy/lib/g2o.cpython-36m-x86_64-linux-gnu.so`?

JeffreyYH, 2021-03-21:
Thanks for the kind explanation, Luigi.

I pulled the new changes and remove my previous g2opy and g2o(cpp) installation and reinstalled third_party by running `./install_thirdparty.sh `and `./install_cpp.sh`. But the problem remains.

The reason that I did not use the script install_all.sh directly is that I need different versions of some python dependencies (e,g. pytorch). But I don't think that is a issue, right?

luigifreda, 2021-03-22:
Do you have rsync installed? 
Did you run the script `install_basic.sh`? 

Please, make sure that the actual commands are executed
[image]

JeffreyYH, 2021-03-23:
Yes, I run install_all.sh and the problem remains.

luigifreda, 2021-03-23:
Could you please answer the following questions? 

1. Do you have another installation of g2o for python?
If yes, could you please try to modify its base folder path temporarily and launch main_slam.py again just to check if there is an interference with the lib paths?

2. Could you please make sure that the following commands are executed 
https://user-images.githubusercontent.com/10080262/112065377-d61ac080-8b64-11eb-920b-b1b9ab3c65e4.png

closed 2021-03-24 by luigifreda

diametrical, 2021-04-13:
Hi Luigi,

I did not use the install_all.sh at first and download all the deps by myself. After I found this thread, I ran the 
./clean.sh and  ran the $ ./install_all.sh. However, I still have the same issue.
module 'g2o' has no attribute 'Flag'.

following outputs after running 
  rsync ./g2opy_changes/types_six_dof_expmap.h ./g2opy/python/types/sba/types_six_dof_expmap.h

Output is 

rsync: change_dir "/home/diam/Downloads/pyslam-master/g2opy//./g2opy_changes" failed: No such file or directory (2)
rsync: change_dir#3 "/home/diam/Downloads/pyslam-master/g2opy//./g2opy/python/types/sba" failed: No such file or directory (2)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(712) [Receiver=3.1.1]

diametrical, 2021-04-13:
Got the same issue, as I installed the g2opy by myself with the wrong branch.

Issued solved by running rsync and make a python3 setup.py install 

    G2OPY_REVISION=5587024
    git checkout $G2OPY_REVISION
    rsync ./g2opy_changes/types_six_dof_expmap.h ./g2opy/python/types/sba/types_six_dof_expmap.h
    rsync ./g2opy_changes/sparse_optimizer.h ./g2opy/python/core/sparse_optimizer.h   
    rsync ./g2opy_changes/python_CMakeLists.txt ./g2opy/python/CMakeLists.txt   
    cd build
    cmake .. $EXTERNAL_OPTION
    make -j8
    cd ..
    python3 setup.py install