luigifreda/pyslam #46: Issue building `orbslam2_features` inside venv

queyrusi, 2021-04-13:
Hi Luigi,

I followed the Mac installation steps on Big Sur 11.2.3 with `install_all_mac_venv.sh` and everything went okay except for third party `orbslam2_features`. The following message comes up:
```
currently used pyenv: pyslam 
-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:19 (message):
  OpenCV > 2.4.3 not found.
```
It would seem cmake doesn't find OpenCV but I can successfully run 
`python3 -c "import cv2; print(cv2.__version__)"` 
to get `3.4.2`.
[This issue](https://github.com/raulmur/ORB_SLAM2/issues/527) seems related but I couldn't successfully apply them.

queyrusi, 2021-04-14:
`CMakeLists.txt` in `./thirdparty/orbslam2_features` doesn't seem to locate any `opencv` although its Python counterpart is well in place (likely under `.python/venvs/pyslam/lib/python3.6/site-packages/cv2`).
Installing it with 
```
brew install opencv@3
```
(since `brew install opencv` tries to fetch opencv4 through a dysfunctional URL somehow so it ends up failing) and linking it in `pyslam/thirdparty/orbslam2_features/CMakeLists.txt`, line 15, through
```
# finds freshly brew-installed opencv
find_package(OpenCV REQUIRED PATHS "/usr/local/opt/opencv@3")
```
instead of
```
find_package(OpenCV ${OPENCV_VERSION} QUIET)
```
fixes the issue.
Any `opencv` related issues regarding building (notably `pyslam_utils` in `./cpp`) can be patched with this workaround. The `CMakeLists.txt` will proceed but launching `python3 -O main_slam.py` will produce 
```
Class CaptureDelegate is implemented in both /Users/username/.python/venvs/pyslam/lib/python3.6/site-packages/cv2/cv2.cpython-36m-darwin.so (0x119e26590) and /usr/local/opt/opencv@3/lib/libopencv_videoio.3.4.dylib (0x13947cfe0). One of the two will be used. Which one is undefined.
```

luigifreda, 2021-04-18:
Thank you very much for your feedback! 
I see the problem is now connected to the fact that 
```
brew install opencv
```
does not work anymore (as you reported above).  I am seeing it on my mac too and this is a bit annoying since it was perfectly working and so I got installed opencv4 sometime ago!

In my setup, if I run (as expected, see the `launch_main_*.sh` files)
```
. pyenv-activate.sh    # required to active the pyslam python env
python3 -c "import cv2; print(cv2.__version__)"
```
I correctly get the `4.5.1` (as I originally expected). 

As for the issue you have with the message 
```
Class CaptureDelegate is implemented in both /Users/username/.python/venvs/pyslam/lib/python3.6/site-packages/cv2/cv2.cpython-36m-darwin.so (0x119e26590) and /usr/local/opt/opencv@3/lib/libopencv_videoio.3.4.dylib (0x13947cfe0). One of the two will be used. Which one is undefined.
```
I am aware of it and I already reported it in the [MAC.md](https://github.com/luigifreda/pyslam/blob/mac/MAC.md) document as an open issue.

Working in order to improve the install procedure.... never-ending story :-)

closed 2022-08-13 by luigifreda