luigifreda/pyslam #49: lfnet missing

JimPlayboy, 2021-04-14:
it seems that the url of lfnet is no longer valid, how can i find a pretrained model of lfnet

JimPlayboy, 2021-04-15:
https://cs.ubc.ca/research/kmyi_data/files/2018/lf-net/lfnet-norotaug.tar.gz
this is a valid url of lfnet offered by Kwang Moo Yi from UBC

closed 2021-04-15 by JimPlayboy

luigifreda, 2021-04-18:
If you try to simply run `wget https://gfx.uvic.ca/pubs/2018/ono2018lfnet/lfnet-norotaug.tar.gz` it still works!

JimPlayboy, 2021-04-18:
no, i don't think so. have you tried to extract the content? the sze of the downloaded file is 2.3kb, much smaller than the alternative one Kwang Moo Yi offered. when i extracted the content, i failed and i was told it is not an archive

luigifreda, 2021-04-18:
you're right, I didn' try to extract it. Ok, I will update that link. Thanks again.