luigifreda/pyslam #50: some problem with pangolin

JimPlayboy, 2021-04-14:
i am working with ubuntu 20.04
i have successfully executed the following commands:
- git clone --recursive https://github.com/luigifreda/pyslam.git
- cd pyslam
- git checkout ubuntu20  
- . pyenv-create.sh
- . install_all_venv.sh  

the command 'python3 -O main_vo.py' works well, but the command 'python3 -O main_slam.py' doesn't work:

(pyslam) (base) leequer@leequer-PC:~/pyslam-ubuntu20$ python3 -O main_slam.py
importing pygame
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
initialising pygame
Traceback (most recent call last):
  File "main_slam.py", line 41, in <module>
    from viewer3D import Viewer3D
  File "/home/leequer/pyslam-ubuntu20/viewer3D.py", line 25, in <module>
    import pypangolin as pangolin
ImportError: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end

it seems that there is something wrong with my pangolin. what should i do？

JimPlayboy, 2021-04-15:
hhh, i have solved this problem by referencing this blog: [Ubuntu20.04安装Pangolin（视觉SLAM十四讲）](https://blog.csdn.net/qq_49331639/article/details/115024110)

1.clean the directory pangolin in thirdparty
2.download the zip of pangolin from https://github.com/stevenlovegrove/Pangolin
3.unzip and copy the content to /thirdparty/pangolin
4.open the terminal under the directory /thirdparty/pangolin , type the following command:
>sudo apt-get install cmake
>sudo apt-get install libglew-dev
>sudo apt-get install libboost-dev libboost-thread-dev libboost-filesystem-dev 
>sudo apt-get install g++
>mkdir build
>cd build
>cmake ..
>make -j2
>sudo make install

5.go to the main directory and edit the file viewer3D.py , replace the line25:
>import pypangolin as pangolin

with
>import pangolin

6.open the terminal under the main directory and type:
>python3 -O main_slam.py

JimPlayboy, 2021-04-15:
the problem still exists, so i removed my solution

i thought i worked because the video and the graph showed up just like main-vo.py

but then i realized that the rebuilt sparse point map was not there. with a closer check, i find the following log information:
>Process Process-1:
Traceback (most recent call last):
  File "/home/leequer/anaconda3/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/leequer/anaconda3/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/leequer/pyslam-ubuntu20/viewer3D.py", line 86, in viewer_thread
    self.viewer_init(kViewportWidth, kViewportHeight)
  File "/home/leequer/pyslam-ubuntu20/viewer3D.py", line 94, in viewer_init
    pangolin.CreateWindowAndBind('Map Viewer', w, h)
AttributeError: module 'pangolin' has no attribute 'CreateWindowAndBind'

and when i repeated the process of make and cmake of pangolin, in the log there was a piece of warning that i missed before, it was something like:
>libpng16.so.16 in the /usr/lib may be hidden by the file in ~/anaconda/lib
you may meet a problem of missing library

and i tried to import pangolin or pypangolin in the console, it showed that "module not found"

so i guess there is a conflict with my anaconda

now i try to install the pyslam using "install_all_conda" and try to understand the usage of CMakeList.txt

JimPlayboy, 2021-04-15:
i found there are some differences between branch master and branch ubuntu20.04:

1.in file install_thirdparty.sh:
* in the branch ubuntu20.04, the pangolin was downloaded from https://gitlab.com/luigifreda/pypangolin (line 92)
* in the branch master, the pangolin was downloaded from https://github.com/uoip/pangolin.git (line 82) and the CMakeList from pangolin_changes was used (line 91)
* both branches use the same make and cmake procedure and generate module 'pangolin' (line 81) because INSTALL_PANGOLIN_ORIGINAL=0 (line 58)
* if i set INSTALL_PANGOLIN_ORIGINAL=1, it should generate module 'pypangolin' (line 61)

2.but in the file viewer3D.py  
* in the branch ubuntu20, it is "import pypangolin as pangolin" in line 25
* while in the branch master, it is "import pangolin" in line 25

should i set INSTALL_PANGOLIN_ORIGINAL=1, or should i change the line 25 in viewer3D.py?

i tried to set INSTALL_PANGOLIN_ORIGINAL=1, delete the /thirdparty/pangolin, and rebuild the pangolin, the problem remains

i tried to quit the conda env (base) and rebuild pangolin, but somehow, after run the command "install_all_venv", the (base) comes back

with the command 
>ln -s pypangolin.*-linux-gnu.so  pangolin.linux-gnu.so

it seems that in file viewer3D.py , “import pangolin”(line 25) is enough for both INSTALL_PANGOLIN_ORIGINAL=1 and INSTALL_PANGOLIN_ORIGINAL=0

but there is a tiny mistake in file install_thirdparty.sh (line 77), the command:
>cd build/src

should be replaced by
>cd src

because we were under the directory \thirdparty/pangolin/build

JimPlayboy, 2021-04-15:
conda is good. nothing has to be changed. 

but i have to make sure every git was complete. because the condition of my network is poor. the failure of git led to most of my problems.

closed 2021-04-15 by JimPlayboy

luigifreda, 2021-04-18:
I've just tested it again under docker and Ubuntu 20 and everything works fine. I followed the instructions here: 
https://github.com/luigifreda/pyslam/blob/ubuntu20/PYTHON-VIRTUAL-ENVS.md

JimPlayboy, 2021-04-18:
@luigifreda when i tried under ubuntu20 using virtual-envs, i came up with following warnings:
>CMake Warning at src/CMakeLists.txt:633 (add_library):
  Cannot generate a safe runtime search path for target pangolin because
  files in some directories may conflict with libraries in implicit
  directories:

>    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/leequer/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/leequer/anaconda3/lib
    runtime library [libtiff.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/leequer/anaconda3/lib
    runtime library [libzstd.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/leequer/anaconda3/lib
    runtime library [liblz4.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/leequer/anaconda3/lib

>  Some of these libraries may not be found correctly.

have you met with these warnings? i think there may be some conflict between cmake and conda envs, but i don't know how to fix it

luigifreda, 2021-04-18:
I don't have conda enabled by default. Probably, the reason is that you have enabled it by default and it's somehow generating some compilation problems. I would suggest not enabling conda by default.

gh279760559, 2021-12-15:
> @luigifreda when i tried under ubuntu20 using virtual-envs, i came up with following warnings:
>
>
>
>
> have you met with these warnings? i think there may be some conflict between cmake and conda envs, [...]

Hi, do you have any problem when installing pangolin, I change INSTALL_PANGOLIN_ORIGINAL=1 because I want to install pypangolin, but could not make it it said 
`common.h:111:10: fatal error: Python.h: No such file or directory`
But I am sure I have python at local place. Cheers

luigifreda, 2021-12-18:
Hi, don't use INSTALL_PANGOLIN_ORIGINAL=1. I will remove it soon cause it's not fully supported in pyslam (I did not test the recent versions of the original pangolin too). 

In any case, those cmake issues with Python.h may be connected to the following cmake calls that may fail 
```
  find_package(PythonInterp) 
  find_package(PythonLibs)
```