luigifreda/pyslam #51: testing under Windows

luigifreda, 2021-04-18:
Some people asked me about how to install and test pyslam under Windows. 
I would suggest going with Windows WSL2 and install ubuntu under it. 
Didn't test it yet but I found this cool install procedure that should provide an ubuntu GUI under Windows too
https://www.youtube.com/watch?v=IL7Jd9rjgrM

SionHu, 2021-05-27:
Got this repo to work flawlessly with WSL2 under Windows 10!! Thank you for the great documentation!

luigifreda, 2021-05-27:
Awesome! Thanks for sharing your experience!

SionHu, 2021-05-27:
No problem!! Just so you know, there was one hiccup that I encountered during the building of g2opy regarding the Eigen quaternion member access. I followed[ this pull request ](https://github.com/uoip/g2opy/pull/52) of g2opy official repo and got it to work. Maybe you can include this in your README in case anyone else has similar problems. 😄

luigifreda, 2021-05-30:
That hiccup is really strange! I've managed that problem here:

luigifreda, 2021-05-30:
I've just checked both under docker and with a real Ubuntu 20 install that g2opy get compiled correctly under the `ubuntu20` branch.

closed 2021-12-11 by luigifreda

luigifreda, 2022-03-18:
Hi,
you can't use `main_vo.py` on TUM datasets. 
This is well explained in the REAMDE 
[image]

Indeed, with  `main_vo.py`

*you need to use a ground truth in order to recover a correct inter-frame scale s and estimate a valid trajectory by composing $C_k = C_{k-1} * [R_{k-1,k}, s t_{k-1,k}]$.*
**This script is a first start to understand the basics of inter-frame feature tracking and camera pose estimation.**

You need the full `main_slam.py` in order to run other datasets.

zxl152723, 2022-03-18:
1

zxl152723, 2022-03-18:
> Hi, you can't use `main_vo.py` on TUM datasets. This is well explained in the REAMDE [image]
>
> Indeed, with `main_vo.py`
>
> _you need to use a ground truth in order to recover a correct inter-frame [...]

Thank you so much for your guidance, this pySLAM project of yours is really nice.

zxl152723, 2022-03-20:
when I run main_slam.py&nbsp; &nbsp;The following error occurs&nbsp;
Traceback (most recent call last):
&nbsp; File "main_slam.py", line 29, in <module&gt;
&nbsp; &nbsp; from slam import Slam, SlamState
&nbsp; File "/mnt/d/pyslam-master/slam.py", line 35, in <module&gt;
&nbsp; &nbsp; from map import Map
&nbsp; File "/mnt/d/pyslam-master/map.py", line 41, in <module&gt;
&nbsp; &nbsp; import optimizer_g2o
&nbsp; File "/mnt/d/pyslam-master/optimizer_g2o.py", line 36, in <module&gt;
&nbsp; &nbsp; def bundle_adjustment(keyframes, points, local_window, fixed_points=False, verbose=False, rounds=10, use_robust_kernel=False, abort_flag = g2o.Flag()):
AttributeError: module 'g2o' has no attribute 'Flag'

zxl152723, 2022-03-20:
when I run main_slam    The following error 

[image]

zxl152723, 2022-03-21:
Hello, I used main_slam.py to run the TUM dataset, but I only want the function of visual odometry, no need to map, how should I change it in the pySLAM program? Or what features are turned off?

Raman-RISE, 2023-03-12:
> I run pySLAM under win10, and only run the visual odometer that is main_vo.py. The trajectory error on the Kitti data set is very small, as shown in the first figure, but the results are basically incorrect when running the TUM data set, as shown in the second as shown in the figure. I have changed the camera parameters in the program, why is this? I have tried several TUM datasets and they are all wrong~ I can't find where the problem is~ [image] [image]

Is it possible to run it directly on Win10 without WSL?

Raman-RISE, 2023-03-12:
> I run pySLAM under win10, and only run the visual odometer that is main_vo.py. The trajectory error on the Kitti data set is very small, as shown in the first figure, but the results are basically [...]

Is it possible to run it on Win10 directly without WSL? How did you install it?