luigifreda/pyslam #53: Support for ScanNet?

SionHu, 2021-05-27:
Hi, there! Thank you so much for your repo! I have everything set up under Windows10 WSL2 and it's working flawlessly! I'm currently working on a research project related to 6DoF camera pose tracking and I found you're repo very useful for some Non-DNN Visual Odometry methods (feature extraction & comparison). 

I'm wondering is adding the dataset support for the [Scannet](http://www.scan-net.org/) part of your plan? It's a very nice dataset providing a lot of information data for similar research tasks.

luigifreda, 2021-05-30:
Thanks for your posts! 
You are very welcome if you want to add support for Scannet! You can start from the code in this file: https://github.com/luigifreda/pyslam/blob/master/dataset.py
I see in Scannet you can basically export the sensor images and intrinsic by using this python code 
https://github.com/ScanNet/ScanNet/tree/master/SensReader/python
Then, you need 
1) to change `config.ini` (in the root folder of pyslam repo) in order to make the code use the `FolderDataset` class from  `dataset.py`: in `config.ini`, change the `[DATASET]` type and update the fields in the `[FOLDER_DATASET]` section
2) transform the exported intrinsics in a file like [this](https://github.com/luigifreda/pyslam/blob/master/settings/WEBCAM.yaml) and change the `cam_settings` field in the above mentioned `[FOLDER_DATASET]` section

luigifreda, 2021-05-30:
Which are your favourite DNN Visual Odometry methods? Can you list them here?

closed 2021-06-05 by luigifreda