luigifreda/pyslam #58: Getting GDK-ERROR

GeorvityLabs, 2021-08-04:
hi folks,

I've ran installed the basic .sh files and while running the vo python file , i'm getting the following error :

(main_vo.py:153596): Gdk-ERROR **: 05:27:39.152: The program 'main_vo.py' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadGC (invalid GC parameter)'.
  (Details: serial 790 error_code 13 request_code 60 (core protocol) minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

GeorvityLabs, 2021-08-04:
Processing Video Input
num frames:  1101
fps:  10.0
using groundtruth:  video
num_levels:  3
using opencv  4.5.3
 WARNING: using NON-ORIENTED features:  SHI_TOMASI - NONE  (i.e. kp.angle=0)
num_levels:  3
 LkFeatureTracker: num levels on LK pyr optic flow:  3
..................................
frame:  0
detector: SHI_TOMASI , #features: 2000 , [kp-filter: SAT ]
The X11 connection broke: No error (code 0)
XIO:  fatal IO error 2 (No such file or directory) on X server ":0"
      after 734 requests (734 known processed) with 14 events remaining

GeorvityLabs, 2021-08-05:
[image]

GeorvityLabs, 2021-08-05:
[image]

luigifreda, 2021-08-06:
Hi,
thanks for the feedback. Could you please share some info about your OS and its configuration? Python version?etc

luigifreda, 2021-08-07:
Another question: how did you install pyslam? Did you use one of the install scripts?

luigifreda, 2021-08-07:
This seems an issue connected to the usage of Tigervnc or similar tools. 
See also this thread https://github.com/gnuradio/gnuradio/issues/3568 
I recommend finding a support forum for TigerVNC and asking there

closed 2021-08-07 by luigifreda

GeorvityLabs, 2021-08-07:
> Hi,
> thanks for the feedback. Could you please share some info about your OS and its configuration? Python version?etc

It's Ubuntu 20.04 , these are the steps I followed :

1) $ git clone --recursive https://github.com/luigifreda/pyslam.git
2) $ git checkout ubuntu20  
3) ./install_basic.sh (which built successfully)
4) python3 -O main_vo.py

This is when the error occured.

luigifreda, 2021-08-08:
Is it an actual Ubuntu 20 installation or it's running under Windows with WSL2?

luigifreda, 2021-08-08:
Just tested under Ubuntu 20.04 (after an upgrade with `sudo apt-get dist-upgrade`)  and under an Ubuntu 20.04 instance run from docker from Ubuntu 18.04 and I didn't get any trouble. 

Did you read the instructions in the README? 
https://github.com/luigifreda/pyslam#install-pyslam-under-ubuntu-2004
You have to follow the instructions on the following page 
https://github.com/luigifreda/pyslam/blob/master/PYTHON-VIRTUAL-ENVS.md 
Do not run 
```
./install_basic.sh
``` 
Instead, run 
```
. install_all_venv.sh
``` 
N.B.: do not forget the dot! (without '/' !)

GeorvityLabs, 2021-08-09:
It's working now , thank you!
I have an .mp4 file I want to test it with, as I only have the .mp4 file and no camera settings or groundtruth file, is the a way i could go about to run the mvo on my .mp4 file.
Any tips to do the same would be greatly appreciated.

I'm familiar with splitting the mp4 file into frames using : ffmpeg and then have a py script to generate timestamps for each image so as to use with ORB SLAM. 
I used this blog post for reference : http://20sep1995.blogspot.com/2019/02/how-to-run-orb-slam-with-your-own-data.html

Would love to do something similar using pyslam.

luigifreda, 2021-08-09:
Did you take a look at the main README? 
https://github.com/luigifreda/pyslam#camera-settings 

If you want to use your camera, you have to:
- calibrate it and configure WEBCAM.yaml accordingly
- record a video (for instance, by using save_video.py in the folder calibration)
- configure the [VIDEO_DATASET] section of config.ini in order to point to your video.