luigifreda/pyslam #60: TypeError: 'NoneType' object is not callable

vivisl, 2021-08-31:
I've built the environment by running the script **install_all.sh** and successfully run **main_vo.py**.
But got TypeError: 'NoneType' object is not callable when running **main_slam.py**.
Need Help! Please check the error below:

```
 WARNING: cannot import ORBextractor from orbslam2_features, check the file TROUBLESHOOTING.md
 WARNING: cannot import pyslam_utils
 WARNING: cannot import Orbslam2Feature2D from feature_orbslam2, check the file TROUBLESHOOTING.md
 WARNING: cannot import DiskFeature2D from feature_disk, check the file TROUBLESHOOTING.md
importing pygame
pygame 2.0.1 (SDL 2.0.14, Python 3.6.13)
Hello from the pygame community. https://www.pygame.org/contribute.html
initialising pygame
Processing Video Input
num frames:  1101
fps:  10.0
tracker_config:  {'num_features': 2000, 'num_levels': 8, 'scale_factor': 1.2, 'detector_type': <FeatureDetectorTypes.ORB2: 7>, 'descriptor_type': <FeatureDescriptorTypes.ORB2: 5>, 'match_ratio_test': 0.7, 'tracker_type': <FeatureTrackerTypes.DES_BF: 1>}
num_levels:  8
using opencv  4.5.3
Traceback (most recent call last):
  File "main_slam.py", line 81, in <module>
    feature_tracker = feature_tracker_factory(**tracker_config)
  File "/media/vivi/E/Documents/Kangyun/Projects/pyslam/feature_tracker.py", line 66, in feature_tracker_factory
    tracker_type = tracker_type)
  File "/media/vivi/E/Documents/Kangyun/Projects/pyslam/feature_tracker.py", line 201, in __init__
    descriptor_type=descriptor_type)                     
  File "/media/vivi/E/Documents/Kangyun/Projects/pyslam/feature_manager.py", line 97, in feature_manager_factory
    return FeatureManager(num_features, num_levels, scale_factor, detector_type, descriptor_type)
  File "/media/vivi/E/Documents/Kangyun/Projects/pyslam/feature_manager.py", line 291, in __init__
    self._feature_detector = Orbslam2Feature2D(self.num_features, self.scale_factor, orb2_num_levels) 
TypeError: 'NoneType' object is not callable

```

closed 2021-08-31 by vivisl

DavidTu21, 2022-03-26:
Hi there, may I ask how did you end up solving this issue? Really appreciate that if there is some hint/clue, thank you in advance for your time.

ochen1, 2022-07-08:
I solved it by building `thirdparty/orbslam2_features`

EthanChou860314, 2023-03-31:
How to build  `thirdparty/orbslam2_features`?
I use ` . install_thirdparty.sh` ,and show Error like this

================================================================== 
Configuring and building thirdparty/orbslam2_features ... 
...
/code/pyslam/thirdparty/orbslam2_features/pybind11/include/pybind11/detail/common.h:111:10: fatal error: Python.h: No such file or directory
  111 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/orbslam2_features.dir/build.make:63：CMakeFiles/orbslam2_features.dir/orb_extractor.cpp.o] error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:96：CMakeFiles/orbslam2_features.dir/all] error 2
make: *** [Makefile:84：all] error 2

================================================ 
Configuring and building thirdparty/Pangolin ... 
pyslam/thirdparty/pangolin/external/pybind11/include/pybind11/detail/common.h:112:10: fatal error: Python.h: No such file or directory
  112 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.

also  g2o

golbal env : Ubuntu20.04 python3.8
My env : . pyenv-conda-create.sh  opencv-contrib-python ==3.4.2.16

Is that mean pybind11 install error?
How can I fix it?

EthanChou860314, 2023-04-14:
I try the Ubuntu 18.04 and 20.04 have the same problem.
Ubuntu 18.04 python3.6.9  pybind11==2.4.3

Maybe I need to know how to connect the correct python.h to the code?

this is my step(something wrong
1. '. pyenv-create.sh'
2. '. pyenv-activate.sh'
3. '. install_thirdparty.sh'

and show the ERROR
Configuring and building thirdparty/Pangolin ...
Configuring and building thirdparty/g2o ...
./pyslam/thirdparty/g2opy/EXTERNAL/pybind11/include/pybind11/detail/common.h:111:10: fatal error: Python.h: No such file or directory
 #include <Python.h>