luigifreda/pyslam #61: Issue with OpenCV and pip from the regular installation on Ubuntu 18.04

diwakar-ravichandran, 2021-09-08:
Hey @luigifreda,

Thank you for such a comprehensive framework, I am seemingly stuck at the very first step of the process, I am using a system with Ubuntu 18.04 and which has conda installed in it. 

However, when I try to install from the install_all_conda.sh, it completes the process, but it fails to load cv2 and just keeps throwing the error there is no cv2 installed. And if I try to check for 

`pip list`

It basically shows me as though there is no pip while I am getting the right output for 
`which pip`
which points to the pip inside the virtual environment pyslam. I am not sure why this is the case. Would appreciate any help in this direction. Also I would request you to put across all the modules that are necessary for the successful running of main_slam.py and main_vo.py so that even if these files don't work as in my case we know what is necessary to manually compile the list of requirements rather than looking through the entire list of things one by one. 

Thanks in advance!

luigifreda, 2021-09-19:
Hi @diwakar-ravichandran  
I cleaned and installed pyslam from scratch under ubuntu 18.04 and then tested the scripts. Everything was smooth as expected. 
Did you read the instructions? 
https://github.com/luigifreda/pyslam/blob/master/CONDA.md
Since you are not seeing `pip` installed I suspect you didn't activate the installed pyslam environment (that must contain pip). You cannot run the pyslam scripts from outside the pyslam environment. 
All the required packages are contained in the file `requirements-conda.yml` that is used by the scripts.

Question: which is your installed conda version?

diwakar-ravichandran, 2021-09-19:
I was able to solve it, actually the thing I made a mistake of was to run it as ./ instead of . <space> which I was able to fix, then I also faced another issue with the third party where there was an issue with g2o source code, that it needed a change in format where it needed w, x, y, z where I had to change according to this,

https://github.com/uoip/g2opy/issues/35

So, yeah, I am working on pyslam right now.

closed 2021-09-19 by diwakar-ravichandran

luigifreda, 2021-09-19:
Great! That issue was already solved under Ubuntu 20 branch
https://github.com/luigifreda/pyslam/blob/ubuntu20/thirdparty/g2opy_changes/eigen_types.h

diwakar-ravichandran, 2021-09-19:
I faced the same issue with Ubuntu 18.04.

luigifreda, 2021-09-19:
fixing that ;-)