luigifreda/pyslam #63: Error Python.h: Arquivo ou diretório inexistente

Daniel-Santos9, 2021-11-25:
When I run the ./install_all.sh file it gives this error.

```
from /home/daniel/Downloads/pyslam-master/thirdparty/orbslam2_features/orb_extractor.cpp:1:
/home/daniel/Downloads/pyslam-master/thirdparty/orbslam2_features/pybind11/include/pybind11/detail/common.h:111:10: fatal error: Python.h: Arquivo ou diretório inexistente
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.
```

luigifreda, 2021-11-27:
Hi Daniel,
did you try 
```
sudo apt-get install python3-dev
```
Which OS are you using?

closed 2021-12-04 by luigifreda

gh279760559, 2021-12-15:
Hello, I got the same problem. The `python3-dev` packagte is installed actually because when I tried to install it said `0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.`

I have multiple python version under `/usr/bin/` directory. The system is Ubuntu 18.04.6 LTS. 

Please help

luigifreda, 2021-12-18:
Hi, 
I am on Ubuntu 18.04.6 LTS and I can't repro the problem. I've just cleaned and successfully rebuilt everything from scratch with the script `install_all.sh`.  

Did you try to use the procedure here? It creates a new virtual python environment and should work as well. 
https://github.com/luigifreda/pyslam#install-pyslam-in-a-custom-python-virtual-environment