luigifreda/pyslam #70: How to use R2D2?

park-dea, 2022-03-10:
Hi Everybody!!
 i want to use the R2D2 network but i run into problems.

tracker_config:  {'num_features': 2000, 'num_levels': 8, 'scale_factor': 1.2, 'detector_type': <FeatureDetectorTypes.R2D2: 22>, 'descriptor_type': <FeatureDescriptorTypes.R2D2: 26>, 'match_ratio_test': 0.7, 'tracker_type': <FeatureTrackerTypes.LK: 0>}
num_levels:  8
using opencv  4.5.5

Traceback (most recent call last):
  File "main_slam.py", line 82, in <module>
    feature_tracker = feature_tracker_factory(**tracker_config)
  File "/home/park/Documents/pyslam-master/feature_tracker.py", line 58, in feature_tracker_factory
    tracker_type = tracker_type)
  File "/home/park/Documents/pyslam-master/feature_tracker.py", line 150, in __init__
    descriptor_type=descriptor_type)   
  File "/home/park/Documents/pyslam-master/feature_manager.py", line 97, in feature_manager_factory
    return FeatureManager(num_features, num_levels, scale_factor, detector_type, descriptor_type)
  File "/home/park/Documents/pyslam-master/feature_manager.py", line 461, in __init__
    self._feature_detector = R2d2Feature2D(num_features=self.num_features)
TypeError: 'NoneType' object is not callable

Any help or advice would be appreciated, thanks.

farhadinima75, 2022-03-14:
Hi @park-dea 
You need to install basic modules after cloning repo.
Run this bash file:
`install_basic.sh`
if above solution didn't work, let me know.

park-dea, 2022-03-15:
Thank you for the reply.
I tried running the 'install_basic.sh' bash file, bur R2D2 doesn't work. 
I've confirmed that other methods work well.(ORB2,  SURF,   ORB2_HARDNET, ORB2_L2NET)

farhadinima75, 2022-03-15:
That's interesting. All the models works well for me.
I've made some modifications to feature detectors and descriptors in this repo to works correctly (specifically with GPU).

Give me  your Email and and I'll send modified `feature_r2d2.py` file to you.

park-dea, 2022-03-15:
Thanks
I'll test it.
E-mail: yulk1234567@gmail.com

farhadinima75, 2022-03-15:
Sent.

luigifreda, 2022-04-23:
Hi @farhadinima75,
you are very welcome to create a pull request for any improvement you think would be useful for this repo.

luigifreda, 2022-04-23:
It seems this has been solved. 
@park-dea what's your OS and which installation procedure did you use?

closed 2022-04-23 by luigifreda

farhadinima75, 2022-04-23:
Hi @luigifreda 
I'll do that. Why not.
Once again, thanks for your great repo.