luigifreda/pyslam #72: AttributeError: module 'g2o' has no attribute 'Flag'

zxl152723, 2022-03-20:
run main_slam.py

zhang@DESKTOP-NCMOEKG:/mnt/d/pyslam-master$ python main_slam.py
Traceback (most recent call last):
  File "main_slam.py", line 29, in <module>
    from slam import Slam, SlamState
  File "/mnt/d/pyslam-master/slam.py", line 35, in <module>
    from map import Map
  File "/mnt/d/pyslam-master/map.py", line 41, in <module>
    import optimizer_g2o
  File "/mnt/d/pyslam-master/optimizer_g2o.py", line 36, in <module>
    def bundle_adjustment(keyframes, points, local_window, fixed_points=False, verbose=False, rounds=10, use_robust_kernel=False, abort_flag = g2o.Flag()):
AttributeError: module 'g2o' has no attribute 'Flag'

closed 2022-03-24 by zxl152723

boyueluzhipeng, 2022-04-14:
你是怎么解决的，感谢分享下

zahidesatmutlu, 2023-09-11:
It's works on Python 3.6.9 and Ubuntu 18.04 WSL2 Subsystem.

hbonnavaud, 2024-03-05:
Hi @DMUZhangXianglong, how did you solved this issue?