luigifreda/pyslam #73: If I run main_slam.py, but I only want to implement the function of VO, how can I change the program?

zxl152723, 2022-03-24:
If I run main_slam.py, but I only want to implement the function of VO, how can I change the program?

luigifreda, 2022-04-03:
Hi, 
thanks for your feedback. Can you explain what you mean by VO?

closed 2022-04-03 by luigifreda

tjennings-cra, 2022-04-13:
I was curious about this, too. I want to do visual odometry without any mapping, but I also want to track scaling. So `visual_odemetry.py` isn't sufficient because it can't track scaling, but `slam.py` is not real-time.

tjennings-cra, 2022-04-13:
I was curious about this too. I want to do visual odometry while tracking scaling, but I don't care about the mapping. So `visual_odometry.py` won't work because it can't track scaling, but `slam.py` is much slower. I was curious how to make something that tracked scaling while not requiring the overhead of building a full map (if that is even possible).

luigifreda, 2022-04-16:
Thanks for your comments. 
First of all, please, read the main README file and the observations therein. 
Next, I strongly suggest reviewing the literature on monocular VO methods and clarifying to yourself (1) which are the required basic ingredients for VO and (2) which is the difference between VO And SLAM approaches. You'll realize that you need at least a local map (contained in a sliding window enclosing the current camera pose) in order to get a decent estimate of the _Sim(3)_ pose without accumulating a huge relative drift (considering scale, translation, orientation) in a short amount time.