luigifreda/pyslam #76: A problem when run main_slam.py

s8088899, 2022-04-18:
Traceback (most recent call last):
  File "main_slam.py", line 29, in <module>
    from slam import Slam, SlamState
  File "/home/szr/pyslam-master/slam.py", line 35, in <module>
    from map import Map
  File "/home/szr/pyslam-master/map.py", line 41, in <module>
    import optimizer_g2o 
  File "/home/szr/pyslam-master/optimizer_g2o.py", line 36, in <module>
    def bundle_adjustment(keyframes, points, local_window, fixed_points=False, verbose=False, rounds=10, use_robust_kernel=False, abort_flag=g2o.Flag()):
AttributeError: module 'g2o' has no attribute 'Flag'

luigifreda, 2022-04-23:
Hi, 
thanks for your feedback.
Take a look at this [thread](https://github.com/luigifreda/pyslam/issues/43). 
I think you are missing rsync in your system. 
```
sudo apt install rsync
```

closed 2022-04-23 by luigifreda

zahidesatmutlu, 2023-09-11:
It works on Python 3.6.9 and Ubuntu 18.04 WSL2 Subsystem.

zahidesatmutlu, 2023-09-11:
It's works on Python 3.6.9 and Ubuntu 18.04 WSL2 Subsystem.