luigifreda/pyslam #78: Question about PLVS

zhangxuelei86, 2022-06-15:
Dear Luigi Freda，I have seen the demo meshing video of PLVS， can you tell which kind of dense reconstruction method and the meshing method had been used？ I want build a pipeline for spatial mapping and will it in agumented reality. What's the new updates of PLVS，as its a great job. Any answers will be great apperciated for your helps!
Yours, Xuelei Zhang

luigifreda, 2022-08-13:
Thanks for your interest. There are different volumetric mapping methods available there (TSDF, 3D occupancy mapping, etc, and some variants of these algorithms). I have to find the time (during my free time) to clean up the repo and add documentation.

closed 2022-08-13 by luigifreda

luigifreda, 2023-10-16:
Hi,
PLVS is now available on github
https://github.com/luigifreda/plvs