luigifreda/pyslam #83: Why use RANSAC and not USAC from open cv?

oscarfossey, 2022-10-10:
Hello thanks for the super repo !

According to this article (https://opencv.org/evaluating-opencvs-new-ransacs/ ) from open CV, USAC performs better and faster than RANSAC do you plan to switch to USAC?

Thank you

Mechazo11, 2022-10-10:
@oscarfossey Can this be easily used in a C++ code i.e. with ORB_SLAM2 ??

oscarfossey, 2022-10-12:
I dont know I never worked directly with ORB_SLAM2 . But every python function from OpenCV as an equivalent in C++ I guess, so maybe.

luigifreda, 2023-10-14:
Hi,
thank you for the feedback. I agree with you. pyslam started some years ago when such RANSAC-based variants were not available under all opencv versions (or at least the most widespread). I will update soon the code by enforcing the usage of the latest opencv versions and will include the USAC algorithm for sure. 

Thanks, 
Luigi

closed 2023-10-14 by luigifreda