luigifreda/pyslam #92: Error with tensorflow-gpu

EduardoQD, 2023-03-31:
Hi,

I'm trying to install this vslam i a docker container but during the instalation I found this error.

ERROR: No matching distribution found for tensorflow-gpu==1.14.0

I'm not sure if I have to install with an other version or something like that

utkarshh27, 2023-04-01:
When the version of tensorflow-gpu you are attempting to install is incompatible with your present system, the error message "No matching distribution found for tensorflow-gpu==1.14.0" appears. You can try installing tensorflow-gpu in a different version that is compatible with your system.

You can alternatively install Tensorflow without GPU support by leaving out the -gpu parameter when configuring the Python environment. As an example, consider the following command:  
`$ export TF_BINARY_URL=https://storage.googleapis.com/tensorflow/mac/cpu/tensorflow-1.14.0-py3-none-any.whl 
$ pip3 install – upgrade $TF_BINARY_URL`

luigifreda, 2023-10-08:
Hi Utkarsh,
thanks for the fix. It works on my side!

closed 2023-11-25 by luigifreda