luigifreda/pyslam #93: ORB_SLAM3

YJonmo, 2023-07-25:
Hi Luigi,

Thanks a lot for this great work.

Is there any Python implementation for the ORB_SLAM3 anywhere?

Regards,
Jacob

FrancescoRossiAiko, 2023-07-25:
about ORB_SLAM3:
- original repo: https://github.com/UZ-SLAMLab/ORB_SLAM3
- python repo github search: https://github.com/search?q=ORB_SLAM3++language%3APython&type=repositories&l=Python

YJonmo, 2023-08-15:
Hi again Luigi,

I am planning to marry your repository with the [PTZ_SLAM](https://github.com/lulufa390/Pan-tilt-zoom-SLAM/tree/master/slam_system). 
I am dealing with stationary lens cameras (rotation and zoom only). The first step that I can think of is to incorporate the [PTZ camera model](https://github.com/lulufa390/Pan-tilt-zoom-SLAM/blob/master/slam_system/ptz_camera.py) into your repository.

The second step is to use deep learning model feature matching such as [LightGlue](https://github.com/cvg/LightGlue). 

The other steps that I am less sure of are loop closure and relocalizations. 
Are there any libraries that you recommend for loop closure and relocalizations? 
The g2opy seems to be a bit obsolete. I found gtsam is up to date and has python bindings. 

Regards,
Jacob

luigifreda, 2023-10-08:
Hi Jacob,
thanks for your feedback. 

- The links provided above (concerning ORBSLAM3) are interesting. Thanks to Francesco.
- As for loop closure tools, you may want to check https://pypi.org/project/pyDBoW3/ (and there are other similar wrappers). I am sure you can find other similar tools by yourself. 

Please, do cite pyslam if you use it in your work. Let me know how it goes. 

Thanks,
Luigi

closed 2023-10-08 by luigifreda