luigifreda/pyslam #94: feature_detector file

abbaselmas, 2023-09-13:
> You can start playing with the supported local features by taking a look at test/cv/test_feature_detector.py and test/cv/test_feature_matching.py.

Under test/cv folder, there is no test_feature_detector.py file.

Just wonder if this is intentional or just dont need that file anymore?
Thanks for your hard work and repo btw!

luigifreda, 2023-09-17:
Hi, thanks for your feedback. Take a look at this 
https://github.com/luigifreda/pyslam/blob/master/test/cv/test_feature_manager2.py#L54
maybe this is what you are looking for...

abbaselmas, 2023-09-18:
Thanks this is self-explanatory indeed.

maybe remove the "test_feature_detector.py" part from the readme for clearance.

closed 2023-09-18 by abbaselmas