luigifreda/pyslam #95: webcam error

hamidreza07, 2023-09-16:
When running the Python script `main_slam.py` with the following command for webcam mode:

```bash
python3 -O main_slam.py
```

I encountered the following error:

```
Traceback (most recent call last):
  File "main_slam.py", line 115, in <module>
    slam.track(img, img_id, timestamp)  # main SLAM function 
  File "/home/hamid/Desktop/vio/new/2/pyslam/slam.py", line 133, in track
    return self.tracking.track(img, frame_id, timestamp)
  File "/home/hamid/Desktop/vio/new/2/pyslam/slam.py", line 511, in track
    f_cur = Frame(img, self.camera, timestamp=timestamp) 
  File "/home/hamid/Desktop/vio/new/2/pyslam/frame.py", line 281, in __init__
    self.kps = kps_data[:,:2]    
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
```

could you please help me?

luigifreda, 2023-10-08:
Hi,
thanks for your feedback. Can you please share some info about your OS (specific config) and the install procedure you used?

hamidreza07, 2023-10-09:
Hi,

Thank you for your prompt response. I'm currently using Ubuntu 18.04 as my operating system, and I've set up a Conda environment to handle the installation process (I followed the installation procedure outlined in the README file).

1. **Operating System:** Ubuntu 18.04

2. **Environment:** Conda Environment 

3.  **python version:** 3.6.10

Best regards.

luigifreda, 2023-10-14:
Which install procedure did you follow?

closed 2023-10-14 by luigifreda