luigifreda/pyslam #96: MatplotLib Not responding

Sadegh-Kalami, 2023-10-06:
Hey,
thanks for your great job. I really love what you have done.
Do you mind helping me?
I did all u have said in readme and code is running perfectly, but the matplotlib(matched_points_plt.draw) is not working.
matplotlib is not responding!!!
[image]

luigifreda, 2023-10-07:
Could you please share which are your OS and the install procedure you followed?

Sadegh-Kalami, 2023-10-08:
thanks for you're fast response.
I have ubuntu 18.04 on VMware and I did the below steps to install PySlam:
1. $ git clone --recursive https://github.com/luigifreda/pyslam.git
2. and then install_all.sh
3. running python3 -O main_slam.py
if you need anything else to tell me to provide it for you.
thanks for your time and consideration <3

luigifreda, 2023-10-08:
Hi,
I see you are using VMWare. Unfortunately, I can't help with that. Virtual machines may cause tons of troubles with viz tools. I recommend using a native Linux system.

closed 2023-10-08 by luigifreda

luigifreda, 2023-10-08:
In any case, try to use `install_all_venv.sh` instead of `install_all.sh`. In general, the installation in the virtual environment is more robust.

Sadegh-Kalami, 2023-10-10:
Appetiate your help and thanks for your support! <3