luigifreda/pyslam #100: main_slam.py 3D map view missing

StrohmayerJ, 2023-11-08:
I tried running the main_slam.py example with _python3 -O main_slam.py_ and expected to get the output shown in the image below (i.e., a camera view and a 3D map view showing the estimated trajectory). The example runs without errors, however, only the camera view window is created.

Should _viewer3D.draw_map(slam)_ create the 3D map view, or is the image below from a different example?

[image]

StrohmayerJ, 2023-11-09:
Update:

I found that the problem was caused by pangolin not building correctly. I tried rebuilding but got this error. Is there a solution for this?

`/home/name/Desktop/pyslam/thirdparty/pangolin/src/video/drivers/uvc.cpp: In member function ‘virtual void pangolin::UvcVideo::Start()’:
/home/name/Desktop/pyslam/thirdparty/pangolin/src/video/drivers/uvc.cpp:208:71: error: too few arguments to function ‘uvc_error_t uvc_stream_start(uvc_stream_handle_t*, void (*)(uvc_frame*, void*), void*, float, uint8_t)’
         uvc_error_t stream_err = uvc_stream_start(strm_, NULL, this, 0);
                                                                       ^
In file included from /home/name/Desktop/pyslam/thirdparty/pangolin/include/pangolin/video/drivers/uvc.h:42:0,
                 from /home/name/Desktop/pyslam/thirdparty/pangolin/src/video/drivers/uvc.cpp:29:
/usr/local/include/libuvc/libuvc.h:561:13: note: declared here
 uvc_error_t uvc_stream_start(uvc_stream_handle_t *strmh,
src/CMakeFiles/_pangolin.dir/build.make:1790: recipe for target 'src/CMakeFiles/_pangolin.dir/video/drivers/uvc.cpp.o' failed
make[2]: *** [src/CMakeFiles/_pangolin.dir/video/drivers/uvc.cpp.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/_pangolin.dir/all' failed
make[1]: *** [src/CMakeFiles/_pangolin.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2`

StrohmayerJ, 2023-11-09:
The problem was apparently caused by a libuvc version mismatch. It worked after installing the latest libuvc version with:

```
git clone https://github.com/libuvc/libuvc
cd libuvc
mkdir build
cd build
cmake ..
make && sudo make install
```

closed 2023-11-09 by StrohmayerJ

shashankyld, 2024-08-20:
I had pip version of pypangolin and it was causing the issue with specific attributes. I uninstalled and it worked out fine.