luigifreda/pyslam #102: Timeseries of camera pose

JackCaster, 2023-12-15:
Hi, I would like to extract the timeseries of the camera pose from SLAM. Specifically, I care about the (relative) rotation of the camera through the video I supply. After a quick investigation, I think the data I need is stored in 
```python
poses = slam.tracking.tracking_history.relative_frame_poses
``` 
and then I can extract each quaternion from  
```python
poses[0].rotation().coeffs()
```
For example, if I want the timeseries of the camera rotation angle (as Euler angles in x, y, z), then I can do
```python
from scipy.spatial.transform import Rotation as R
rots = [R.from_matrix(pose.R).as_euler('xyz', degrees=True) for pose in poses]
rots = np.asarray(rots).cumsum(axis=0)
```
Would this be a correct approach?

luigifreda, 2023-12-15:
Hi,
this was already asked in issue #97 

https://github.com/luigifreda/pyslam/issues/97#issuecomment-1762851078

closed 2023-12-15 by luigifreda

JackCaster, 2023-12-18:
Thank you. In that comment you mentioned how to retrieve the current pose, but you also said that the final estimate may differ. How does one get the final timeseries of poses? [Here](https://github.com/luigifreda/pyslam/issues/84#issuecomment-1762854063) you write that the function to store the final data is still work in progress---is it still the case?