luigifreda/pyslam #105: Solving ImportError: ../pyslam/thirdparty/g2opy/lib/g2o.so: undefined symbol: _Py_ZeroStruct  |   Again...

nl22-nmsu, 2024-01-19:
Hey, I love how well your instructions are laid out for this repo, I'm excited to use it! But I'm running into an issue that someone else had, only the posted solution isn't working for me. 

Here's the rundown:
I'm using WSL 2 with ubuntu 20.04
I have numpy 1.22.1 ( I have also tried 1.24.4 but get an error about typeDict )
And I checked out the ubuntu20 version

As posted in #79 I replaced `cmake .. $EXTERNAL_OPTION` with `cmake .. -DPYTHON_EXECUTABLE=/usr/bin/python3 ` in line 133 of the install_thirdparty.sh file but the error still persists.

I made sure to run `./clean.sh` and then `. install_all.sh` to make sure that it was a clean build too. 

I also noticed that it always installs numpy version 1.24.4 during the build, so I'm not sure if the downgrade is what's causing this or something else. 

I'd love any help on what could be causing this error and how to solve it

luigifreda, 2024-01-19:
Hi,
which install procedure did you use?

nl22-nmsu, 2024-01-19:
I followed the ubuntu 20 install.

The exact commands I ran:
```
git clone --recursive https://github.com/luigifreda/pyslam.git
cd pyslam 
git checkout ubuntu20
. install_all.sh
```

There was a slight modification that I made to utils_goem.py in the poseRT() function, instead of `ret[:3,3] = t` I changed it to `ret[:3, 3] = np.array(t).reshape(3,1).squeeze()` because there was an odd type error or something like that. Though this wouldn't cause an issue from what I can tell.

nl22-nmsu, 2024-01-20:
Okay, I figured it out.

There were a few things that were building incorrectly. I made 3 changes:

1) as previously stated, I modified line 133 in `install_thirdparty.sh`

2) I also had the same _Py_ZeroStruct error for pangolin. To solve that I added two cmake tags to two lines. I'm not too sure what the difference is for the installs so I just added these tags to both based on [this](https://github.com/uoip/pangolin/issues/33) issue I found.
     line 74: `cmake ../ -DAVFORMAT_INCLUDE_DIR="" -DCPP11_NO_BOOST=ON $EXTERNAL_OPTION -DBUILD_PANGOLIN_FFMPEG=OFF -DPYBIND11_PYTHON_VERSION=3.8`

    line 101: `cmake .. -DBUILD_PANGOLIN_LIBREALSENSE=OFF $EXTERNAL_OPTION -DBUILD_PANGOLIN_FFMPEG=OFF -DPYBIND11_PYTHON_VERSION=3.8 # disable realsense `

3) Once I got past these issues I had an issue building `thirdparty/orbslam2_features`. I had the `TypeError: 'NoneType' object is not callable` from feature_manager. To resolve this I changed the cmake command  in `thirdparty/orbslam2_features/build.sh` to `cmake .. -DPYTHON_EXECUTABLE=/usr/bin/python3`. Then did a clean build on that folder.

Now things seem to be working. Hope this helps anyone else who may have these same issues.

closed 2024-01-20 by nl22-nmsu

luigifreda, 2024-01-20:
Hi @nl22-nmsu,
thank you for your feedback. 

Unfortunately, you installed pyslam with a wrong procedure. 
With ubuntu 20 or ubuntu 22, you need to follow the instructions reported in this section:
https://github.com/luigifreda/pyslam?tab=readme-ov-file#ubuntu-2004-and-ubuntu-2204
In particular, once you run 
```
$ git clone --recursive https://github.com/luigifreda/pyslam.git
$ cd pyslam 
$ git checkout ubuntu20  
```
you have to follow the instructions on this page:
https://github.com/luigifreda/pyslam/blob/master/PYTHON-VIRTUAL-ENVS.md
In particular, you need to run 
```
$ . install_all_venv.sh 
```
and NOT
```
$ . install_all.sh 
```
as you did.

nl22-nmsu, 2024-01-20:
Oh, I see, my impressions was that the virtual environment was optional. I'll redo it with the virtual environment to make sure everything is running correctly. It seemed to be working on my round about way of installing from what I can tell though. 

Thanks for your prompt replies!