luigifreda/pyslam #113: AttributeError: module 'g2o' has no attribute 'Flag'

shashankyld, 2024-08-17:
(pyslam) shashank@legion:~/Documents/UniBonn/Sem4/ThesisPrep/pyslam$ python main_slam.py 
Traceback (most recent call last):
  File "main_slam.py", line 29, in <module>
    from slam import Slam, SlamState
  File "/home/shashank/Documents/UniBonn/Sem4/ThesisPrep/pyslam/slam.py", line 35, in <module>
    from map import Map
  File "/home/shashank/Documents/UniBonn/Sem4/ThesisPrep/pyslam/map.py", line 41, in <module>
    import optimizer_g2o 
  File "/home/shashank/Documents/UniBonn/Sem4/ThesisPrep/pyslam/optimizer_g2o.py", line 36, in <module>
    def bundle_adjustment(keyframes, points, local_window, fixed_points=False, verbose=False, rounds=10, use_robust_kernel=False, abort_flag=g2o.Flag()):
AttributeError: module 'g2o' has no attribute 'Flag':

I used the correct branch for my OS (Ubuntu 20) and I am using conda.

tlaurie99, 2024-08-19:
If you just want to get the code working, you can set all of these to `False` since it seems like it is a flag for earlier termination based on the pyslam/test/g2o/test_optimization_flag.py file. I also get this same error.

luigifreda, 2024-08-19:
Thanks for your feedback. 
This error happens when the customized g2o does not succeed in getting built and imported by pyslam code.

I pushed some fixes/improvements for the conda build/install procedure. 
Can you try running the following commands from the root folder of your working copy?
```bash
git pull origin master   
./clean.sh
./install_all_conda.sh 
```
Then follow the other instructions here: 
https://github.com/luigifreda/pyslam/blob/master/CONDA.md

tlaurie99, 2024-08-20:
@luigifreda as a follow up from my side. My issue was a little different -- I was building a docker image with a few other repos included and noticed I was not building the correct g2o. I was using g2o-python. As soon as I used the correct g2opy this worked well.

closed 2024-08-23 by luigifreda