luigifreda/pyslam #117: Failed to build h5py

TheIllusionist77, 2024-09-14:
When running this line to install pySLAM on Raspberry Pi OS (using the Ubuntu installation instructions),

`./install_all_venv.sh`

I get the following errors:

`ERROR: Failed building wheel for h5py`
`Failed to build h5py`
`ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (h5py)`

Is there any solution? Help is much appreciated.

luigifreda, 2024-09-15:
Hi,
thanks for your feedback. 

Your problem should be related to this post 
https://github.com/h5py/h5py/issues/2035#issuecomment-1009183432

Add the following line in the script `install_system_packages.sh` 
```
install_package libhdf5-dev    # needed when building h5py wheel from src is required (arm64)
```
and let me know how it goes.

TheIllusionist77, 2024-09-15:
This resolved the issue. Thank you!

closed 2024-09-15 by TheIllusionist77