luigifreda/pyslam #123: error while Computing groundtruth associations for euroc and tum datasets

aimiliosnrobotics, 2024-12-11:
Hello everyone and thank you very much for the nice work. Im trying to test the package with some dataset that includes depth information (stereo/rgbd) to check out the Volumetric reconstruction but currently i am getting the same error: 
```
if abs(float(a[0])  - (float(b[0])  + offset)) < max_difference]
ValueError: could not convert string to float: '#timestamp,'
```
with full terminal output for euroc and tum datasets respectively: 
```
Processing Euroc groundtruth of lenght:  36383
 Computing groundtruth associations (one-time operation)...
Traceback (most recent call last):
  File "./main_slam.py", line 70, in <module>
    groundtruth = groundtruth_factory(config.dataset_settings)
  File "/home/aimpet/pyslam/ground_truth.py", line 65, in groundtruth_factory
    return EurocGroundTruth(path, name, associations, start_frame_id, GroundTruthType.EUROC)
  File "/home/aimpet/pyslam/ground_truth.py", line 465, in __init__
    self.association_matches = self.associate(self.image_data, self.data)
  File "/home/aimpet/pyslam/ground_truth.py", line 520, in associate
    potential_matches = [(abs(float(a[0]) - (float(b[0]) + offset)), ia, ib) # a[0] and b[0] extract the first element which is a timestamp 
  File "/home/aimpet/pyslam/ground_truth.py", line 523, in <listcomp>
    if abs(float(a[0])  - (float(b[0])  + offset)) < max_difference]
IndexError: list index out of range
```
```
Processing TUM Sequence
using groundtruth:  tum
base_path:  /home/aimpet/pyslam/../../aimpet/tum/rgbd_dataset_freiburg2_pioneer_360
 Computing groundtruth associations (one-time operation)...
Traceback (most recent call last):
  File "./main_slam.py", line 70, in <module>
    groundtruth = groundtruth_factory(config.dataset_settings)
  File "/home/aimpet/pyslam/ground_truth.py", line 63, in groundtruth_factory
    return TumGroundTruth(path, name, associations, start_frame_id, GroundTruthType.TUM)
  File "/home/aimpet/pyslam/ground_truth.py", line 339, in __init__
    self.association_matches = self.associate(self.associations, self.data)
  File "/home/aimpet/pyslam/ground_truth.py", line 414, in associate
    potential_matches = [(abs(float(a[0]) - (float(b[0]) + offset)), ia, ib) # a[0] and b[0] extract the first element which is a timestamp 
  File "/home/aimpet/pyslam/ground_truth.py", line 417, in <listcomp>
    if abs(float(a[0])  - (float(b[0])  + offset)) < max_difference]
IndexError: list index out of range
```

when trying to run                         ./main_slam.py, both with the euroc as well as with the tum dataset. (didnt try the kitti dataset yet because i see that i need to subscribe when im trying downloading it from the provided link). I created a virtual environment pyslam with venv following the steps for ubuntu 20.04 running ./install_all_venv.sh . Thank you very much in advance!

luigifreda, 2024-12-11:
You may want to pull to get updated documentation and a stricter check in `EurocGroundTruth` class. 
https://github.com/luigifreda/pyslam/blob/master/ground_truth.py#L445

aimiliosnrobotics, 2024-12-11:
Thank you very much for the fast response. I was able to run it with the euroc stereo dataset after generating correctly the groundtruths file and using the new check in the EurocGroundTruth class. Thank you!

closed 2024-12-11 by aimiliosnrobotics