luigifreda/pyslam #128: error for "./main_vo.py"

JiePaiWs, 2024-12-24:
Thank you for your amazing work.However,I get some problem.The error are as follows, i wonder why:

2024-12-24 22:15:40.607881: I tensorflow/core/util/port.cc:110] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-12-24 22:15:40.644143: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
 WARNING: cannot import ALIKED from lightglue, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import SIFT from lightglue, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import LightGlue from lightglue, check the file docs/TROUBLESHOOTING.md
 INFO: rerun running
Traceback (most recent call last):
  File "./main_vo.py", line 56, in <module>
    from viewer3D import Viewer3D
  File "/home/seu/jh_ws/pyslam/viz/viewer3D.py", line 28, in <module>
    import pypangolin as pangolin
ImportError: /home/seu/jh_ws/pyslam/thirdparty/pangolin/pypangolin.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZTIN8pangolin9ImageViewE

luigifreda, 2024-12-24:
Hi,
Thank you for the feedback.

Can you share what your OS is and what installation procedure you run? Are you running the script from the pyslam environment?
Did you read the README.md file?

JiePaiWs, 2024-12-24:
sure,my OS is ubuntu20.04 and i have run the conda installation.I catch some trouble in the Thirdpart installation too.

---- Replied Message ----
| From | Luigi ***@***.***> |
| Date | 12/25/2024 00:04 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [luigifreda/pyslam] error for "./main_vo.py" (Issue #128) |

Hi,
Thank you for the feedback.

Can you share what your OS is and what installation procedure you run? Are you running the script from the pyslam environment?
Did you read the README.md file?

luigifreda, 2024-12-24:
Can you share the errors you got during the installation procedure? 
On my side, I am checking back the conda installation under a clean docker ubuntu20 with conda.

JiePaiWs, 2024-12-24:
ok,thanks again for your amazing work.I will organize the errors I have encountered soon.And good night for you.It is time for sleep haha

---- Replied Message ----
| From | Luigi ***@***.***> |
| Date | 12/25/2024 00:14 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [luigifreda/pyslam] error for "./main_vo.py" (Issue #128) |

Can you share the errors you got during the installation procedure?
On my side, I am checking back the conda installation under a clean docker ubuntu20 with conda.

JiePaiWs, 2024-12-25:
Hello,the organization errors are as follows:
when i run ". install_thirdparty.sh":
================================================================== 
Configuring and building thirdparty/pyibow ... 
EXTERNAL_OPTIONS: 

Configuring and building thirdparty/pyibow/obindex2 ...
-- Build type: Release
-- Setting Release options
-- Build with -march=native
-- Compiler flags:  -Wall -O3 -fPIC -DNDEBUG -march=native -fopenmp
-- Configuring done
-- Generating done
-- Build files have been written to: /home/seu/jh_ws/pyslam/thirdparty/pyibow/modules/obindex2/lib/build
Consolidate compiler generated dependencies of target obindex2
[ 38%] Built target obindex2
Consolidate compiler generated dependencies of target test_bdesc
Consolidate compiler generated dependencies of target test_btree
Consolidate compiler generated dependencies of target test_bindex
Consolidate compiler generated dependencies of target test_search
[ 53%] Built target test_bindex
[ 61%] Linking CXX executable test_search
[ 76%] Built target test_btree
[ 92%] Built target test_bdesc
/usr/bin/ld: CMakeFiles/test_search.dir/tests/test_search.cc.o: in function `void std::__introsort_loop<__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, long, __gnu_cxx::__ops::_Iter_less_iter>(__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, __gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, long, __gnu_cxx::__ops::_Iter_less_iter) [clone .isra.0]':
test_search.cc:(.text+0x76): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: test_search.cc:(.text+0x89): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: test_search.cc:(.text+0x9c): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: test_search.cc:(.text+0xcc): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: test_search.cc:(.text+0xee): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: CMakeFiles/test_search.dir/tests/test_search.cc.o:test_search.cc:(.text+0x14c): more undefined references to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const' follow
/usr/bin/ld: CMakeFiles/test_search.dir/tests/test_search.cc.o: in function `getFilenames(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*)':
test_search.cc:(.text+0x5ff): undefined reference to `boost::filesystem::path::extension_v3() const'
/usr/bin/ld: CMakeFiles/test_search.dir/tests/test_search.cc.o: in function `void std::__unguarded_linear_insert<__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, __gnu_cxx::__ops::_Val_less_iter>(__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, __gnu_cxx::__ops::_Val_less_iter)':
test_search.cc:(.text._ZSt25__unguarded_linear_insertIN9__gnu_cxx17__normal_iteratorIPN5boost10filesystem4pathESt6vectorIS4_SaIS4_EEEENS0_5__ops14_Val_less_iterEEvT_T0_[_ZSt25__unguarded_linear_insertIN9__gnu_cxx17__normal_iteratorIPN5boost10filesystem4pathESt6vectorIS4_SaIS4_EEEENS0_5__ops14_Val_less_iterEEvT_T0_]+0xaa): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: CMakeFiles/test_search.dir/tests/test_search.cc.o: in function `void std::__insertion_sort<__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, __gnu_cxx::__ops::_Iter_less_iter>(__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, __gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, __gnu_cxx::__ops::_Iter_less_iter)':
test_search.cc:(.text._ZSt16__insertion_sortIN9__gnu_cxx17__normal_iteratorIPN5boost10filesystem4pathESt6vectorIS4_SaIS4_EEEENS0_5__ops15_Iter_less_iterEEvT_SC_T0_[_ZSt16__insertion_sortIN9__gnu_cxx17__normal_iteratorIPN5boost10filesystem4pathESt6vectorIS4_SaIS4_EEEENS0_5__ops15_Iter_less_iterEEvT_SC_T0_]+0x65): undefined reference to `boost::filesystem::path::compare_v3(boost::filesystem::path const&) const'
/usr/bin/ld: CMakeFiles/test_search.dir/tests/test_search.cc.o: in function `void std::__adjust_heap<__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, long, boost::filesystem::path, __gnu_cxx::__ops::_Iter_less_iter>(__gnu_cxx::__normal_iterator<boost::filesystem::path*, std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > >, long, long, boost::filesystem::path, __gnu_cxx::__ops::_Iter_less_iter)':
[... cut ...]
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/seu/jh_ws/pyslam/thirdparty
-----------------------------------------------------
besides ,i found the other errors and i finshed it.
when i run . install_thirdparty.sh  ,there is errors about opencv.I changed the "find_package(OpenCV  QUIET)"  to "ind_package(OpenCV 4 QUIET)".The errors are done.I think it is on account of the version of OpenCV.There are OPENCV 3.4.10 and 4.2.0 in my OS.

JiePaiWs, 2024-12-25:
and when i run bash: ". main_vo.py"
* This file is part of PYSLAM 
*
* Copyright (C) 2016-present Luigi Freda <luigi dot freda at gmail dot com> 
*
* PYSLAM is free software: you can redistribute it and/or modify
* 
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.
*
* PYSLAM is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU General Public License
* along with PYSLAM. If not, see <http://www.gnu.org/licenses/>.
: 没有那个文件或目录
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/413.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/413.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/413.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/413.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/413.
from: can't read /var/mail/config
from: can't read /var/mail/visual_odometry
from: can't read /var/mail/camera
from: can't read /var/mail/ground_truth
from: can't read /var/mail/dataset
from: can't read /var/mail/mplot_thread
from: can't read /var/mail/qplot_thread
from: can't read /var/mail/feature_tracker
from: can't read /var/mail/feature_tracker_configs
from: can't read /var/mail/rerun_interface
kUseRerun：未找到命令
bash: main_vo.py: 行 46: 未预期的符号“(”附近有语法错误
bash: main_vo.py: 行 46: `if kUseRerun and not Rerun.is_ok():'

luigifreda, 2024-12-25:
> and when i run bash: ". main_vo.py"

You are supposed to run 
`./main_vo.py` 
and not 
`. main_vo.py`

JiePaiWs, 2024-12-25:
oh,sorry.I run the ./main_vo.py and errors are as follows:
 WARNING: cannot import SuperPointFeature2D from feature_superpoint, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import XFeat2D from feature_xfeat, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import TfeatFeature2D from feature_tfeat, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import HardnetFeature2D from feature_hardnet, check the file docs/TROUBLESHOOTING.md
2024-12-25 17:00:03.794951: I tensorflow/core/util/port.cc:110] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-12-25 17:00:03.971926: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
 WARNING: cannot import GeodescFeature2D from feature_geodesc, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import SosnetFeature2D from feature_sosnet, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import L2NetFeature2D from feature_l2net, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import LogpolarFeature2D from feature_logpolar, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import D2NetFeature2D from feature_d2net, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import DelfFeature2D from feature_delf, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import ContextDescFeature2D from feature_contextdesc, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import LfNetFeature2D from feature_lfnet, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import R2d2Feature2D from feature_r2d2, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import KeyNetDescFeature2D from feature_keynet, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import DiskFeature2D from feature_disk, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import AlikedFeature2D from feature_aliked, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import LightGlueSIFTFeature2D from feature_lightglue_sift, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import KeyNetAffNetHardNetFeature2D from feature_keynet_affnet_hardnet, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "./main_vo.py", line 29, in <module>
    from visual_odometry import VisualOdometry
  File "/home/seu/jh_ws/pyslam/slam/visual_odometry.py", line 25, in <module>
    from feature_tracker import FeatureTrackerTypes, FeatureTrackingResult, FeatureTracker
  File "/home/seu/jh_ws/pyslam/local_features/feature_tracker.py", line 25, in <module>
    from feature_matcher import feature_matcher_factory, FeatureMatcherTypes
  File "/home/seu/jh_ws/pyslam/local_features/feature_matcher.py", line 23, in <module>
    import torch
  File "/home/seu/anaconda3/envs/pyslam/lib/python3.8/site-packages/torch/__init__.py", line 235, in <module>
    from torch._C import *  # noqa: F403
ImportError: libcupti.so.12: cannot open shared object file: No such file or directory

i wonder how i can fix it

luigifreda, 2024-12-25:
Did you read the README file?

If you plan to use Conda, please refer to the dedicated document at:
https://github.com/luigifreda/pyslam/blob/master/docs/CONDA.md

The errors you reported suggest that the Boost libraries are missing. However, the boost libraries are installed automatically if you run the following command:
`./install_all_conda.sh`

After the installation process is complete, you need to activate the pyslam environment by running:
`. pyenv-conda-activate.sh`

Once the environment is activated, you can execute the main script. Everything should work as expected.

Yesterday, I tested the Conda installation procedure using Docker ([rosdocker](https://github.com/luigifreda/rosdocker) with the `ubuntu_conda` image), and everything ran smoothly.

JiePaiWs, 2024-12-25:
Thanks for your request, I'd like to know the specific command line to install Boost. I think the issue might be related to the version of Boost I have installed. My current Boost version is 1.75, and I will try to uninstall the existing version and reinstall it.

luigifreda, 2024-12-25:
The install procedure will take care of installing the boost system library  with:
`sudo apt install -y libboost-serialization-dev libboost-system-dev libboost-filesystem-dev`

If you have a specific configuration on your system, I recommend using the docker procedure (see the README).
The install scripts were created to perform all the required install operations for you and make the installation process as painless as possible.
If you want to install things manually instead of using the install scripts, follow the same steps as the install scripts. I won't be able to support that.
Good luck!

closed 2024-12-25 by luigifreda