luigifreda/pyslam #131: Problem when relocalizing

AlbertoJaenal, 2025-01-15:
Hi Luigi!

Thanks for considering my suggestions in the previous issue. Is quite nice to see how the repo improves constantly!

I have found another issue that may be a corner case. I am running the pipeline in a quite challenging dataset, and I want to relocalize continously in a map created with a previous sequence. I have the problem that in some sequences, it relocalizes and then gets lost quickly (5-6 frames), and then the slam gets stuck in a "None" mode.
I found that it is because this condition: https://github.com/luigifreda/pyslam/blob/master/slam/tracking.py#L1019
So, as it is a relocalizing session, it resets the whole slam (including the map), and it cannot relocalize again because the map gets reset and then it gets stuck. 
Maybe here, in the relocalization case, it is better to say `self.map.num_keyframes() <= 5`. Or maybe not to reset the map?

luigifreda, 2025-01-15:
Hi Alberto,
Thank you very much for your feedback, and suggestions and for "stressing" pySLAM with cool tests. 

I see your point - good catch! 
From an initial check, I believe a quick and dirty solution is to comment out `self.loop_closing.request_reset()` in `slam.reset_session()` [here](https://github.com/luigifreda/pyslam/blob/master/slam/slam.py#L160). This would result in:
```python
    def reset_session(self):
        self.local_mapping.request_reset()
        #if self.loop_closing is not None:
        #   self.loop_closing.request_reset()
        if self.volumetric_integrator is not None:
            self.volumetric_integrator.request_reset()  
        self.tracking.reset()
        self.map.reset_session()  
``` 
Indeed, `self.map.reset_session()` correctly manages the map reset (without removing reloaded data). However, resetting the `loop_closing` makes all loop-closure submodules unusable. 

Can you make a couple of tests for me with your datasets and see if this works? 

BTW, if you'd like to make a pull request from your fork (in case you're not familiar with this GitHub workflow), here's a useful guide
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

LMK

AlbertoJaenal, 2025-01-16:
Hi, Luigi

I tried the patch and it worked. I did not dare to create a pull request because my knowledge of the pipeline is still shallow.
Thanks for your answer!

closed 2025-01-16 by AlbertoJaenal

luigifreda, 2025-01-16:
Thank you very much for your suggestions and for testing pySLAM!