luigifreda/pyslam #132: AttributeError: module 'pydbow3' has no attribute 'BowVector'

shahaabshokouhi, 2025-01-24:
I am trying to run the code but I am getting this error. 
acl_laptop@acllaptop:~/Documents/Codes/pyslam$ /bin/python3 /home/acl_laptop/Documents/Codes/pyslam/main_slam.py
 WARNING: cannot import LightGlue from lightglue, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import ORBextractor from orbslam2_features, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import pyslam_utils
 WARNING: cannot import TfeatFeature2D from feature_tfeat, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import Orbslam2Feature2D from feature_orbslam2, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import GeodescFeature2D from feature_geodesc, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import LogpolarFeature2D from feature_logpolar, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import D2NetFeature2D from feature_d2net, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import DelfFeature2D from feature_delf, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import ContextDescFeature2D from feature_contextdesc, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import LfNetFeature2D from feature_lfnet, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import R2d2Feature2D from feature_r2d2, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import KeyNetDescFeature2D from feature_keynet, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import DiskFeature2D from feature_disk, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import ALIKED from lightglue, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import SIFT from lightglue, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import KeyNetAffNetHardNetFeature2D from feature_keynet_affnet_hardnet, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "/home/acl_laptop/Documents/Codes/pyslam/main_slam.py", line 30, in <module>
    from slam import Slam, SlamState
  File "/home/acl_laptop/Documents/Codes/pyslam/slam/slam.py", line 41, in <module>
    from loop_closing import LoopClosing
  File "/home/acl_laptop/Documents/Codes/pyslam/loop_closing/loop_closing.py", line 40, in <module>
    from loop_detector_configs import LoopDetectorConfigs
  File "/home/acl_laptop/Documents/Codes/pyslam/loop_closing/loop_detector_configs.py", line 37, in <module>
    from loop_detector_dbow3 import LoopDetectorDBoW3
  File "/home/acl_laptop/Documents/Codes/pyslam/loop_closing/loop_detector_dbow3.py", line 69, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase): 
  File "/home/acl_laptop/Documents/Codes/pyslam/loop_closing/loop_detector_dbow3.py", line 107, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5): 
AttributeError: module 'pydbow3' has no attribute 'BowVector'

luigifreda, 2025-01-24:
Hi,
Could you share the installation procedure you followed?
Also, what operating system are you using?
Have you had a chance to review the README file?

closed 2025-01-24 by luigifreda

shahaabshokouhi, 2025-01-24:
Hi Luigi
Thanks for your response. I followed the instruction exactly. I used conda and venv both but each one has its own errors. I am using linux. The steps I followed are:

clean.sh
install_all_conda.sh
activate the env
main_slam.py

Do I need to do anything else?

luigifreda, 2025-01-24:
Could you please reply to my questions. What is your OS? Do you have cuda installed?

Let's try again. 

First, check that you cloned the repo recursively. 

Next, run exactly the following steps. Open a shell and get into the repo root. Then run:
```
clean.sh --hard
./install_all_venv.sh
.  pyenv-activate     
./main_slam.py 
```
Please, copy paste in a txt file all the std output of the install process and share it here. 

LMK.
Luigi