luigifreda/pyslam #133: cv2 problem on ubuntu 24

ufukasia, 2025-01-28:
Hello i am using ubuntu24 

For installation, I followed the instructions from [[this page](https://github.com/luigifreda/pyslam/blob/master/docs/CONDA.md)](https://github.com/luigifreda/pyslam/blob/master/docs/CONDA.md).  
I ran the following commands:

```bash
$ ./install_all_conda.sh
$ . pyenv-conda-activate.sh
$ ./main_vo.py
```

However, when I run `./main_vo.py`, I get the following error:

```bash
(pyslam) masa24@masa24-MS-7C56:~/Desktop/pyslam$ ./main_vo.py
Traceback (most recent call last):
  File "./main_vo.py", line 22, in <module>
    import cv2
  File "/home/masa24/anaconda3/envs/pyslam/lib/python3.8/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/home/masa24/anaconda3/envs/pyslam/lib/python3.8/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/home/masa24/anaconda3/envs/pyslam/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/masa24/anaconda3/envs/pyslam/bin/../lib/libgcc_s.so.1: version `GCC_12.0.0' not found (required by /lib/x86_64-linux-gnu/libhwy.so.1)
```

I also tried reinstalling OpenCV using the provided shell script, but I still encounter the same issue.

Could you please help me identify the problem and guide me on how to resolve this? Thank you!

ufukasia, 2025-01-28:
### Solution:
I replaced the outdated `libgcc_s.so.1` file in the Anaconda environment with the correct version from the system libraries.

1. **Backup the Existing File in Anaconda**  
   To avoid losing the original file, I made a backup of the outdated version:
   ```bash
   mv ~/anaconda3/envs/pyslam/lib/libgcc_s.so.1 ~/anaconda3/envs/pyslam/lib/libgcc_s.so.1.bak
   ```

2. **Copy the Correct Version from the System**  
   I located the correct version of `libgcc_s.so.1` in the system library directory (usually `/usr/lib/x86_64-linux-gnu/`) and copied it into the Anaconda environment:
   ```bash
   cp /usr/lib/x86_64-linux-gnu/libgcc_s.so.1 ~/anaconda3/envs/pyslam/lib/
   ```

3. **Run the Program Again**  
   After replacing the file, I ran the program, and the issue was resolved.

---

### Why This Worked:
The Anaconda environment often uses its own versions of system libraries for isolation and compatibility. However, when a library in the environment is outdated or mismatched, it can cause errors. By replacing the problematic file with the updated system version, the program could access the necessary library features without errors.

SOLVED😊

luigifreda, 2025-01-28:
Hi,
Thank you very much for sharing your solution. I'll share it in the troubleshooting document
https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md

Best,
Luigi

closed 2025-01-28 by luigifreda