luigifreda/pyslam #134: failed creating the venv when using ./install_all_venv.sh

nusbaum-uriel, 2025-02-04:
when running this shell command it ended with (on ubuntu 22.04)

Building wheels for collected packages: kiwisolver, psutil, wrapt
  Building wheel for kiwisolver (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [11 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'kiwisolver' extension
      creating build/temp.linux-x86_64-cpython-38/py
      clang++ -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DOPENSSL_NO_SSL3 -fPIC -I. -I/home/uriel/.python/venvs/pyslam/include -I/home/uriel/.pyenv/versions/3.8.10/include/python3.8 -c py/constraint.cpp -o build/temp.linux-x86_64-cpython-38/py/constraint.o
      py/constraint.cpp:8:10: fatal error: 'algorithm' file not found
      #include <algorithm>
               ^~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang++' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for kiwisolver
  Running setup.py clean for kiwisolver
  Building wheel for psutil (setup.py) ... done
  Created wheel for psutil: filename=psutil-5.5.1-cp38-cp38-linux_x86_64.whl size=249851 sha256=8f57ea4cf65aba0c50cecc231092380d11c708cc03ecf11d1a56d0413dfb357d
  Stored in directory: /home/uriel/.cache/pip/wheels/57/f5/41/32f5a1f2b1012be55d33a36be92cd310ec0fe44c3fdc042783
  Building wheel for wrapt (setup.py) ... done
  Created wheel for wrapt: filename=wrapt-1.11.2-cp38-cp38-linux_x86_64.whl size=60469 sha256=61e0bc9ad74da5a51fe69ce6d8c9366641107e8139deebd5bc5fce7a368b8c57
  Stored in directory: /home/uriel/.cache/pip/wheels/52/25/16/7e228f52ee876e9c9c66dfd6eaf6fb2f73f18629a74dcf7045
Successfully built psutil wrapt
Failed to build kiwisolver
ERROR: Failed to build installable wheels for some pyproject.toml based projects (kiwisolver)

luigifreda, 2025-02-04:
I'm double-checking the procedure.

It looks like the build process for the `kiwisolver` package depends on clang++. In the meantime, could you verify whether libstdc++-11-dev is installed on your system or if you're using a different version?"

nusbaum-uriel, 2025-02-05:
running "sudo apt-get -y install libstdc++-11-dev" yields:

   Reading package lists... Done
   Building dependency tree... Done
   Reading state information... Done
   libstdc++-11-dev is already the newest version (11.4.0-1ubuntu1~22.04).

so the installed version is 11.4.0-1

luigifreda, 2025-02-05:
I am unsure how to help since I cannot replicate your problem. 
Could you share more info about your system and chipset?
- OS
- `gcc --version`
- CUDA version if installed 

Can you please open a new shell, get into the root of pyslam and then run the following commands? 
```
./clean.sh --hard 
./install_all_venv.sh
```
After, can you please copy-paste all the std output you obtained into a text file and share it here?

Can you also try to use [rosdocker](https://github.com/luigifreda/rosdocker) and the ubuntu22_cuda container?

luigifreda, 2025-02-05:
This seems to be a related issue
https://github.com/immunant/c2rust/issues/380
They suggest installing `libstdc++-12-dev` as a workaround under ubuntu22.  However, that's something I cannot verify on my side.

closed 2025-02-07 by luigifreda