luigifreda/pyslam #139: Errors with venv-based installation on Ubuntu 22

saeejithnair, 2025-02-18:
Hi, I tried running `smnair@guacamole:/pub0/smnair/robotics/pyslam$ ./install_all_venv.sh ` on an Ubuntu 22 machine and ran into the following errors. Previously I tried to build the conda environment and ran into a different issue (https://github.com/luigifreda/pyslam/issues/138)

```
Installing collected packages: ruff
Successfully installed ruff-0.9.6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cuda-command-line-tools-12-6 is already the newest version (12.6.3-1).
cuda-libraries-12-6 is already the newest version (12.6.3-1).
The following packages were automatically installed and are no longer required:
  libdw-dev libelf-dev libgudev-1.0-dev liborc-0.4-dev liborc-0.4-dev-bin libwpe-1.0-1 libwpebackend-fdo-1.0-1 libx11-xcb-dev nsight-compute-2024.3.1
  nsight-systems-2024.4.2 nvidia-firmware-535-535.183.01
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 239 not upgraded.
Building with 16 cores
-- The CUDA compiler identification is unknown
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - failed
-- Check for working CUDA compiler: /usr/bin/nvcc
-- Check for working CUDA compiler: /usr/bin/nvcc - broken
CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCUDACompiler.cmake:56 (message):
  The CUDA compiler

    "/usr/bin/nvcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /pub0/smnair/robotics/pyslam/thirdparty/lietorch/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/ninja cmTC_8dcd9 && [1/2] Building CUDA object CMakeFiles/cmTC_8dcd9.dir/main.cu.o
    FAILED: CMakeFiles/cmTC_8dcd9.dir/main.cu.o 
    /usr/bin/nvcc      -c /pub0/smnair/robotics/pyslam/thirdparty/lietorch/build/CMakeFiles/CMakeTmp/main.cu -o CMakeFiles/cmTC_8dcd9.dir/main.cu.o
    /usr/include/c++/11/type_traits(1406): error: type name is not allowed
    
    /usr/include/c++/11/type_traits(1406): error: type name is not allowed
    
    /usr/include/c++/11/type_traits(1406): error: identifier "__is_same" is undefined
    
    /usr/include/c++/11/type_traits(3251): error: type name is not allowed
    
    /usr/include/c++/11/type_traits(3251): error: type name is not allowed
    
    /usr/include/c++/11/bits/stl_pair.h(460): error: argument list for class template "std::pair" is missing
    
    /usr/include/c++/11/bits/stl_pair.h(460): error: expected a ")"
    
    /usr/include/c++/11/bits/stl_pair.h(460): error: template parameter "_T1" may not be redeclared in this scope
    
    /usr/include/c++/11/bits/stl_pair.h(460): error: expected a ";"
    
    9 errors detected in the compilation of "/pub0/smnair/robotics/pyslam/thirdparty/lietorch/build/CMakeFiles/CMakeTmp/main.cu".
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/pub0/smnair/robotics/pyslam/thirdparty/lietorch/build/CMakeFiles/CMakeOutput.log".
See also "/pub0/smnair/robotics/pyslam/thirdparty/lietorch/build/CMakeFiles/CMakeError.log".
ninja: error: loading 'build.ninja': No such file or directory
ninja: error: loading 'build.ninja': No such file or directory
Processing ./thirdparty/monogs/submodules/simple-knn
[... 136 lines cut ...]
  Cloning https://github.com/facebookresearch/pytorch3d.git (to revision stable) to /tmp/pip-req-build-ar0zg96u
  Running command git clone --filter=blob:none --quiet https://github.com/facebookresearch/pytorch3d.git /tmp/pip-req-build-ar0zg96u
  Running command git checkout -q 75ebeeaea0908c5527e7b1e305fbc7681382db47
  Resolved https://github.com/facebookresearch/pytorch3d.git to commit 75ebeeaea0908c5527e7b1e305fbc7681382db47
  Preparing metadata (setup.py) ... done
Collecting iopath (from pytorch3d==0.7.8)
  Using cached iopath-0.1.10-py3-none-any.whl
Requirement already satisfied: tqdm in /home/smnair/.python/venvs/pyslam/lib/python3.10/site-packages (from iopath->pytorch3d==0.7.8) (4.66.4)
Requirement already satisfied: typing-extensions in /home/smnair/.python/venvs/pyslam/lib/python3.10/site-packages (from iopath->pytorch3d==0.7.8) (4.12.2)
Collecting portalocker (from iopath->pytorch3d==0.7.8)
  Downloading portalocker-3.1.1-py3-none-any.whl.metadata (8.6 kB)
Downloading portalocker-3.1.1-py3-none-any.whl (19 kB)
Building wheels for collected packages: pytorch3d
  Building wheel for pytorch3d (setup.py) ... done
  Created wheel for pytorch3d: filename=pytorch3d-0.7.8-cp310-cp310-linux_x86_64.whl size=71815319 sha256=f9c254d1008c4f57a3e70dbeb682b81d2b3940263fe023a5b862a7772da8237b
  Stored in directory: /tmp/pip-ephem-wheel-cache-76pmij38/wheels/39/5f/20/2d3b6f3a35a60bdc0ba3c19da94340db9596637d1d1222473d
Successfully built pytorch3d
Installing collected packages: portalocker, iopath, pytorch3d
Successfully installed iopath-0.1.10 portalocker-3.1.1 pytorch3d-0.7.8
WITH_PYTHON_INTERP_CHECK: ON 
EXTERNAL_OPTIONS:  -DWITH_PYTHON_INTERP_CHECK=ON
Using conda pyslam...

CondaError: Run 'conda init' before 'conda activate'
```

luigifreda, 2025-02-18:
Hi,

Thanks for your feedback. Looking at your log, I see a couple of issues:

1. It appears that the CUDA libraries are either not installed on your system or not properly detected. In particular, I see the following message:
```
Check for working CUDA compiler: /usr/bin/nvcc - broken
```
This suggests that there is a problem with your CUDA configuration. You may need to reinstall or properly configure CUDA to resolve this.

2. The log indicates that conda is still being used in the environment. It looks like conda is set as your default Python environment manager, but mixing it with venv is not recommended. The script `./install_all_venv.sh` is designed for venv and is not compatible with conda.

closed 2025-02-18 by luigifreda

saeejithnair, 2025-02-20:
Hi @luigifreda, thank you for taking a look at the log. Here is my output from running those commands
```
smnair@guacamole:/pub0/smnair/robotics/pyslam$ ./cuda-config.sh
cat /usr/local/cuda/version.txt
nvidia-smi
echo $PATH
echo $LD_LIBRARY_PATH 
bash: ./cuda-config.sh: No such file or directory
[bat error]: '/usr/local/cuda/version.txt': No such file or directory (os error 2)
Thu Feb 20 14:47:56 2025       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.216.01             Driver Version: 535.216.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA RTX A6000               On  | 00000000:41:00.0  On |                  Off |
| 30%   41C    P8              24W / 300W |    340MiB / 49140MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA RTX A6000               On  | 00000000:42:00.0 Off |                  Off |
| 30%   38C    P8              20W / 300W |     10MiB / 49140MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   2  NVIDIA RTX A6000               On  | 00000000:61:00.0 Off |                  Off |
| 30%   40C    P8              23W / 300W |     10MiB / 49140MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      4562      G   /usr/lib/xorg/Xorg                          162MiB |
|    0   N/A  N/A      4769      G   /usr/bin/gnome-shell                         61MiB |
|    0   N/A  N/A      5434      G   ...ures=SpareRendererForSitePerProcess       38MiB |
|    0   N/A  N/A     18813      G   ...88,262144 --variations-seed-version       31MiB |
|    1   N/A  N/A      4562      G   /usr/lib/xorg/Xorg                            4MiB |
|    2   N/A  N/A      4562      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+
/home/smnair/.pyenv/shims:/home/smnair/.pyenv/bin:/home/smnair/.pyenv/bin:/home/smnair/.pixi/bin:/home/smnair/.local/bin:/home/smnair/.cursor-server/cli/servers/Stable-f5f18731406b73244e0558ee7716d77c8096d150/server/bin/remote-cli:/home/smnair/.local/bin:/home/smnair/.pyenv/bin:/home/smnair/.pyenv/bin:/home/smnair/.pixi/bin:/home/smnair/.nvm/versions/node/v18.16.0/bin:/home/smnair/miniconda3/condabin:/home/smnair/.local/bin:/home/smnair/.cargo/bin:/home/smnair/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/bin:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/share/OpenCV:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/bin:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/share/OpenCV:/pub0/smnair/robotics/HSLAM/Thirdparty/CompiledLibs/bin:/pub0/smnair/robotics/HSLAM/Thirdparty/CompiledLibs/share/OpenCV:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/bin:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/share/OpenCV:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/bin:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/share/OpenCV:/pub0/smnair/robotics/HSLAM/Thirdparty/CompiledLibs/bin:/pub0/smnair/robotics/HSLAM/Thirdparty/CompiledLibs/share/OpenCV
:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/lib:/pub0/smnair/robotics/hslam_ros_ws/src/HSLAM/Thirdparty/CompiledLibs/lib:/pub0/smnair/robotics/HSLAM/Thirdparty/CompiledLibs/lib
```

and additionally,
```
smnair@guacamole:/pub0/smnair/robotics/pyslam$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0
```