luigifreda/pyslam #149: Issues in localization

lavishuse1, 2025-02-27:
Hi
I want to know that when a apply pyslam over my own video dataset. Every time the localization fails after some frames and due to which further working stops. As i can see the number of feature it founds are more and always present but the feature it uses (the green circles) vanishes and hence localization fails.
Any specific changes needed or i need to preprocess my video .
Help me with specification.
Waiting for a fast response 
Thanks

lavishuse1, 2025-02-28:
What changes need for loop closing detection and it optimized automatically.
Currently i am using AKAZE and VLAD .

luigifreda, 2025-02-28:
Hi,
Did you read the section [loop closing](https://github.com/luigifreda/pyslam?tab=readme-ov-file#loop-closing) contained in the README?

closed 2025-03-01 by luigifreda